docbender/openHAB-Simatic #1: OpenHAB2 needs several restarts to start polling values, polling stops if items are refreshed

AchilleGR, 2016-12-09:
Hello,

I am testing your binding on an OpenHAB2, connected to a Siemens Logo 8 in S7 mode. Nice work by the way, apart from this issue, your binding seems to work OK!

I am running openhab on a raspberry pi 3 which is somewhat underpowered, so timing issues are easier to reproduce.
I need to restart openhab several times until polling works OK, and if I do any changes in the items file, polling stops altogether. tcpdump shows that no info is being sent to the plc, which resets the connection afterwards.

I think this is an issue with bindingChanged which doesn't refresh the polling list properly. Also, if this happens while polling, you get stuck in a busy-wait loop.

Please tell me if I can be of further assistance in debugging this issue.

[openhab-simatic-items-refresh-issue.txt](https://github.com/docbender/openHAB-Simatic/files/642036/openhab-simatic-items-refresh-issue.txt)
[openhab-simatic-normal-startup.txt](https://github.com/docbender/openHAB-Simatic/files/642037/openhab-simatic-normal-startup.txt)
[openhab-simatic-startup-issue.txt](https://github.com/docbender/openHAB-Simatic/files/642038/openhab-simatic-startup-issue.txt)
[tcpdump.txt](https://github.com/docbender/openHAB-Simatic/files/642039/tcpdump.txt)

docbender, 2016-12-11:
Hello @AchilleGR, 

to be honest, I did not test binding under OH2 and probably not even configuration change on-the-fly. 

I know that there could be a problem with slow responding PLC or network. Perhaps this is behind the startup issue. I will look on it and let you know.

docbender, 2016-12-12:
I tested it under OH2 a I found missing pid in binding manifest and I also add binding activator. Both should help with startup issue. I create new release so u can test it if u want.

Tomorrow I provide more testing and also take a look to the on-the-fly configuration changes.

AchilleGR, 2016-12-12:
Hi @docbender ,

I created a pull request for you, please check it out. Thanks!

commit 455f5715 (2016-12-13): Fix for issue #1 (#3)

* Binding will not start if there isn't a configuration.
Support for modifying the configuration on the fly
Better locking avoids busy-wait loops when changing Items while running.
Fixes #1.

* Comment that binding also works with openHAB 2.0

* Resolved conflict with upstream
closed 2016-12-13 by docbender