docbender/openHAB-Simatic #2: Establish Openhab2 readiness and merge to Openhab officially

shorty707, 2016-12-11:


docbender, 2016-12-13:
Now binding is ready to use in OH2

shorty707, 2016-12-13:
cool.
are you planning full oh2 support?

and merge to openhab that it can installed via paperui etc?

docbender, 2016-12-13:
Right now OH2 is supported in 1.x compatibility mode. About pure OH2 support? I will think about it when stable OH2 version will be released.

Yea installation via paper ui is cool thing. No doubt. And installation without it is a bit of pain, till you find where to copy configuration... But since my last binding pull request passed 11 month till someone began review. So yes installation through paperui is cool, but here is the most recent version.

jogibear9988, 2017-05-06:
As 2.x is released, do you still have plans to work on paperui installation support!

docbender, 2017-05-15:
I had it in plan but now I do not time for it. And as I know there is already S7 binding in review for PaperUI (https://github.com/openhab/openhab2-addons/pull/1606). So there would be no reason to build new one if this one would be merged.

grzegorz914, 2018-01-18:
Hi,
I get communications error if want to read or write to Area above 304. 
Simatic S7 1215C, openHab 2.2/2.3 

> 2018-01-18 11:23:08.172 [ERROR] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - Read data area error (Area=DB19.DBB208-DB19.DBB374, Return code=0xffffff80, Error=Unexpected function code in answer)

any solution for this

grzegorz914, 2018-01-19:
Hi,

thanks for Your help,
Yes I need all bytes in this area (my project is really big)
temporary I have set optional "O" for all Data above 304 and communications work OK but in this state I cannot write data to PLC abowe this memory area. { simatic="plc:DB19.DBD318:float:**O**" }. 
1215C is little different and have much more memory as 1212C. It will be nice if this bindings can be extended to support more memory area.

Bottom tail from my Items:
> Group     PrzedpokojMalyTemp          "Przedpokój mały"  [ "Thermostat" ]
> Number    PLC_PrzedpokojMalyTempA      "Przedpokój mały"  <temperature> (PrzedpokojMalyTemp) [ "CurrentTemperature" ]   { simatic="plc:DB28.DBD2:float" }
> Number    PLC_PrzedpokojMalyTempZ     "Przedpokój mały"  <temperature> (PrzedpokojMalyTemp) [ "TargetTemperature" ]     { simatic="plc:DB19.DBD220:float" }
> String    PLC_PrzedpokojMalyTempTryb "Przedpokój mały" <text> (PrzedpokojMalyTemp) [ "homekit:HeatingCoolingMode" ]     
> Number    PLC_PrzedpokojMalyTermostatStan    "Przedpokój mały"  <radiator> { simatic="plc:DB19.DBB219:byte" }
> 
> Group     PrzedpokojDuzyTemp          "Przedpokój duży"  [ "Thermostat" ]
> Number    PLC_PrzedpokojDuzyTempA      "Przedpokój duży"  <temperature> (PrzedpokojDuzyTemp) [ "CurrentTemperature" ]   { simatic="plc:DB28.DBD6:float" }
> Number    PLC_PrzedpokojDuzyTempZ     "Przedpokój duży"  <temperature> (PrzedpokojDuzyTemp) [ "TargetTemperature" ]     { simatic="plc:DB19.DBD234:float" }
> String    PLC_PrzedpokojDuzyTempTryb "Przedpokój duży" <text> (PrzedpokojDuzyTemp) [ "homekit:HeatingCoolingMode" ]
> Number    PLC_PrzedpokojDuzyTermostatStan    "Przedpokój duży"  <radiator> { simatic="plc:DB19.DBB233:byte" }
> 
> Group     PokojParterTemp          "Pokój parter"  [ "Thermostat" ]
> Number    PLC_PokojParterTempA      "Pokój parter"  <temperature> (PokojParterTemp) [ "CurrentTemperature" ]   { simatic="plc:DB28.DBD10:float" }
> Number    PLC_PokojParterTempZ     "Pokój parter"  <temperature> (PokojParterTemp) [ "TargetTemperature" ]     { simatic="plc:DB19.DBD248:float" }
> String    PLC_PokojParterTempTryb  "Pokoj parter" <text> (PokojParterTemp) [ "homekit:HeatingCoolingMode" ]
> Number   PLC_PokojParterTermostatStan    "Pokoj parter"  <radiator> { simatic="plc:DB19.DBB247:byte" }
> 
> Group     SalonTemp          "Salon"  [ "Thermostat" ]
> Number    PLC_SalonTempA      "Salon"  <temperature> (SalonTemp) [ "CurrentTemperature" ]   { simatic="plc:DB28.DBD14:float" }
> Number    PLC_SalonTempZ     "Salon"  <temperature> (SalonTemp) [ "TargetTemperature" ]     { simatic="plc:DB19.DBD262:float" }
> String    PLC_SalonTempTryb  "Salon" <text> (SalonTemp) [ "homekit:HeatingCoolingMode" ]
> Number    PLC_SalonTermostatStan    "Salon"  <light> { simatic="plc:DB19.DBB261:byte" }
> 
> Group     JadalniaTemp          "Jadalnia"  [ "Thermostat" ]
> Number    PLC_JadalniaTempA      "Jadalnia"  <temperature> (JadalniaTemp) [ "CurrentTemperature" ]   { simatic="plc:DB28.DBD14:float" }
> Number    PLC_JadalniaTempZ     "Jadalnia"  <temperature> (JadalniaTemp) [ "TargetTemperature" ]     { simatic="plc:DB19.DBD276:float" }
> String    PLC_JadalniaTempTryb "Jadalnia" <text> (JadalniaTemp) [ "homekit:HeatingCoolingMode" ]
> Number    PLC_JadalniaTermostatStan    "Jadalnia"  <radiator> { simatic="plc:DB19.DBB261:byte" }
> 
> Group     KuchniaTemp          "Kuchnia"  [ "Thermostat" ]
> Number    PLC_KuchniaTempA      "Kuchnia"  <temperature> (KuchniaTemp) [ "CurrentTemperature" ]   { simatic="plc:DB28.DBD18:float" }
> Number    PLC_KuchniaTempZ     "Kuchnia"  <temperature> (KuchniaTemp) [ "TargetTemperature" ]     { simatic="plc:DB19.DBD290:float" }
> String    PLC_KuchniaTempTryb     "Kuchnia" <text> (KuchniaTemp) [ "homekit:HeatingCoolingMode" ]
> Number    PLC_KuchniaTermostatStan    "Kuchnia"  <radiator> { simatic="plc:DB19.DBB289:byte" }
> 
> Group     LazienkaParterTemp          "Łazienka parter"  [ "Thermostat" ]
> Number    PLC_LazienkaParterTempA      "Łazienka parter"  <temperature> (LazienkaParterTemp) [ "CurrentTemperature" ]   { simatic="plc:DB28.DBD22:float" }
> Number    PLC_LazienkaParterTempZ  "Łazienka parter"  <temperature> (LazienkaParterTemp) [ "TargetTemperature" ]   { simatic="plc:DB19.DBD304:float" }
> String    PLC_LazienkaParterTempTryb  "Łazienka parter" <text> (LazienkaParterTemp) [ "homekit:HeatingCoolingMode" ]
> Number    PLC_LazienkaParterTermostatStan    "Łazienka parter"  <radiator> { simatic="plc:DB19.DBB303:byte" }
> 
> Group     GarderobaTemp          "Garderoba"  [ "Thermostat" ]
> Number    PLC_GarderobaTempA      "Garderoba"  <temperature> (GarderobaTemp) [ "CurrentTemperature" ]   { simatic="plc:DB28.DBD26:float" }
> Number    PLC_GarderobaTempZ     "Garderoba"  <temperature> (GarderobaTemp) [ "TargetTemperature" ]     { simatic="plc:DB19.DBD318:float:O" }
> String    PLC_GarderobaTempTryb  "Garderoba" <text> (GarderobaTemp) [ "homekit:HeatingCoolingMode" ]
> Number    PLC_GarderobaTermostatStan    "Garderoba"  <radiator> { simatic="plc:DB19.DBB317:byte:O" }
> 
> Group     BiuroTemp          "Biuro"  [ "Thermostat" ]
> Number    PLC_BiuroTempA      "Biuro"  <temperature> (BiuroTemp) [ "CurrentTemperature" ]   { simatic="plc:DB28.DBD30:float" }
> Number    PLC_BiuroTempZ     "Biuro"  <temperature> (BiuroTemp) [ "TargetTemperature" ]     { simatic="plc:DB19.DBD332:float:O" }
> String    PLC_BiuroTempTryb  "Biuro" <text> (BiuroTemp) [ "homekit:HeatingCoolingMode" ]
[... cut ...]
> Number    PLC_LazienkaPietroTempA      "Łazienka piętro"  <temperature> (LazienkaPietroTemp) [ "CurrentTemperature" ]   { simatic="plc:DB28.DBD42:float" }
> Number    PLC_LazienkaPietroTempZ     "Łazienka piętro"  <temperature> (LazienkaPietroTemp) [ "TargetTemperature" ]     { simatic="plc:DB19.DBD374:float:O" }
> String    PLC_LazienkaPietroTempTryb  "Łazienka piętro" <text> (LazienkaPietroTemp) [ "homekit:HeatingCoolingMode" ]
> Number    PLC_LazienkaPietroTermostatStan    "Łazienka piętro"  <radiator> { simatic="plc:DB19.DBB373:byte:O" }

Thanks

docbender, 2018-01-19:
Just tested your config and whole area is read:

> 
> 12:47:30.808 [DEBUG] [.b.simatic.internal.SimaticTCP:127  ] - isoTCP:192.168.254.99 - open() - connecting
> 12:47:31.315 [INFO ] [.b.simatic.internal.SimaticTCP:166  ] - isoTCP:192.168.254.99 - connected
> 12:47:31.316 [DEBUG] [.b.simatic.internal.SimaticTCP:337  ] - isoTCP:192.168.254.99 - checkNewData() is called
> 12:47:31.318 [DEBUG] [.b.simatic.internal.SimaticTCP:348  ] - Locking
> 12:47:31.571 [DEBUG] [.b.simatic.internal.SimaticTCP:395  ] - isoTCP:192.168.254.99 - Reading finished. Area=M0-M1
> 12:47:31.572 [DEBUG] [.b.simatic.internal.SimaticTCP:402  ] - isoTCP:192.168.254.99 - PostValue for item=S71212MB0 (Device=plc1212 MemAddress=MB0/MB0 DataType=BYTE Direction=0)
> 12:47:31.576 [DEBUG] [o.o.b.s.i.SimaticGenericDevice:550  ] - isoTCP:192.168.254.99 - Incoming data - item:S71212MB0/state:127
> 12:47:31.830 [DEBUG] [.b.simatic.internal.SimaticTCP:395  ] - isoTCP:192.168.254.99 - Reading finished. Area=DB19.DBB219-DB19.DBB378
> 12:47:31.831 [DEBUG] [.b.simatic.internal.SimaticTCP:402  ] - isoTCP:192.168.254.99 - PostValue for item=PLC_PrzedpokojMalyTermostatStan (Device=plc1212 MemAddress=DB19.DBB219/DB19.DBB219 DataType=BYTE Direction=0)
> 12:47:31.834 [DEBUG] [o.o.b.s.i.SimaticGenericDevice:550  ] - isoTCP:192.168.254.99 - Incoming data - item:PLC_PrzedpokojMalyTermostatStan/state:127
> 12:47:31.836 [DEBUG] [.b.simatic.internal.SimaticTCP:402  ] - isoTCP:192.168.254.99 - PostValue for item=PLC_PrzedpokojMalyTempZ (Device=plc1212 MemAddress=DB19.DBD220/DB19.DBD220 DataType=FLOAT Direction=0)
> 12:47:31.841 [DEBUG] [o.o.b.s.i.SimaticGenericDevice:550  ] - isoTCP:192.168.254.99 - Incoming data - item:PLC_PrzedpokojMalyTempZ/state:0
> 12:47:31.844 [DEBUG] [.b.simatic.internal.SimaticTCP:402  ] - isoTCP:192.168.254.99 - PostValue for item=PLC_PrzedpokojDuzyTermostatStan (Device=plc1212 MemAddress=DB19.DBB233/DB19.DBB233 DataType=BYTE Direction=0)
> 12:47:31.849 [DEBUG] [o.o.b.s.i.SimaticGenericDevice:550  ] - isoTCP:192.168.254.99 - Incoming data - item:PLC_PrzedpokojDuzyTermostatStan/state:0
> 12:47:31.851 [DEBUG] [.b.simatic.internal.SimaticTCP:402  ] - isoTCP:192.168.254.99 - PostValue for item=PLC_PrzedpokojDuzyTempZ (Device=plc1212 MemAddress=DB19.DBD234/DB19.DBD234 DataType=FLOAT Direction=0)
> 12:47:31.855 [DEBUG] [o.o.b.s.i.SimaticGenericDevice:550  ] - isoTCP:192.168.254.99 - Incoming data - item:PLC_PrzedpokojDuzyTempZ/state:0
> 12:47:31.859 [DEBUG] [.b.simatic.internal.SimaticTCP:402  ] - isoTCP:192.168.254.99 - PostValue for item=PLC_PokojParterTermostatStan (Device=plc1212 MemAddress=DB19.DBB247/DB19.DBB247 DataType=BYTE Direction=0)
> 12:47:31.864 [DEBUG] [o.o.b.s.i.SimaticGenericDevice:550  ] - isoTCP:192.168.254.99 - Incoming data - item:PLC_PokojParterTermostatStan/state:0
> 12:47:31.866 [DEBUG] [.b.simatic.internal.SimaticTCP:402  ] - isoTCP:192.168.254.99 - PostValue for item=PLC_PokojParterTempZ (Device=plc1212 MemAddress=DB19.DBD248/DB19.DBD248 DataType=FLOAT Direction=0)
> 12:47:31.871 [DEBUG] [o.o.b.s.i.SimaticGenericDevice:550  ] - isoTCP:192.168.254.99 - Incoming data - item:PLC_PokojParterTempZ/state:0
> 12:47:31.876 [DEBUG] [.b.simatic.internal.SimaticTCP:402  ] - isoTCP:192.168.254.99 - PostValue for item=PLC_JadalniaTermostatStan (Device=plc1212 MemAddress=DB19.DBB261/DB19.DBB261 DataType=BYTE Direction=0)
> 12:47:31.879 [DEBUG] [o.o.b.s.i.SimaticGenericDevice:550  ] - isoTCP:192.168.254.99 - Incoming data - item:PLC_JadalniaTermostatStan/state:0
> 12:47:31.883 [DEBUG] [.b.simatic.internal.SimaticTCP:402  ] - isoTCP:192.168.254.99 - PostValue for item=PLC_SalonTermostatStan (Device=plc1212 MemAddress=DB19.DBB261/DB19.DBB261 DataType=BYTE Direction=0)
> 12:47:31.886 [DEBUG] [o.o.b.s.i.SimaticGenericDevice:550  ] - isoTCP:192.168.254.99 - Incoming data - item:PLC_SalonTermostatStan/state:0
> 12:47:31.891 [DEBUG] [.b.simatic.internal.SimaticTCP:402  ] - isoTCP:192.168.254.99 - PostValue for item=PLC_SalonTempZ (Device=plc1212 MemAddress=DB19.DBD262/DB19.DBD262 DataType=FLOAT Direction=0)
> 12:47:31.896 [DEBUG] [o.o.b.s.i.SimaticGenericDevice:550  ] - isoTCP:192.168.254.99 - Incoming data - item:PLC_SalonTempZ/state:0
> 12:47:31.898 [DEBUG] [.b.simatic.internal.SimaticTCP:402  ] - isoTCP:192.168.254.99 - PostValue for item=PLC_JadalniaTempZ (Device=plc1212 MemAddress=DB19.DBD276/DB19.DBD276 DataType=FLOAT Direction=0)
> 12:47:31.902 [DEBUG] [o.o.b.s.i.SimaticGenericDevice:550  ] - isoTCP:192.168.254.99 - Incoming data - item:PLC_JadalniaTempZ/state:0
> 12:47:31.906 [DEBUG] [.b.simatic.internal.SimaticTCP:402  ] - isoTCP:192.168.254.99 - PostValue for item=PLC_KuchniaTermostatStan (Device=plc1212 MemAddress=DB19.DBB289/DB19.DBB289 DataType=BYTE Direction=0)
> 12:47:31.910 [DEBUG] [o.o.b.s.i.SimaticGenericDevice:550  ] - isoTCP:192.168.254.99 - Incoming data - item:PLC_KuchniaTermostatStan/state:0
> 12:47:31.912 [DEBUG] [.b.simatic.internal.SimaticTCP:402  ] - isoTCP:192.168.254.99 - PostValue for item=PLC_KuchniaTempZ (Device=plc1212 MemAddress=DB19.DBD290/DB19.DBD290 DataType=FLOAT Direction=0)
> 12:47:31.918 [DEBUG] [o.o.b.s.i.SimaticGenericDevice:550  ] - isoTCP:192.168.254.99 - Incoming data - item:PLC_KuchniaTempZ/state:0
> 12:47:31.920 [DEBUG] [.b.simatic.internal.SimaticTCP:402  ] - isoTCP:192.168.254.99 - PostValue for item=S71212DB300 (Device=plc1212 MemAddress=DB19.DBB300/DB19.DBB300 DataType=BYTE Direction=0)
[... cut ...]
> 12:47:32.056 [DEBUG] [o.o.b.s.i.SimaticGenericDevice:550  ] - isoTCP:192.168.254.99 - Incoming data - item:PLC_LazienkaPietroTermostatStan/state:0
> 12:47:32.060 [DEBUG] [.b.simatic.internal.SimaticTCP:402  ] - isoTCP:192.168.254.99 - PostValue for item=PLC_LazienkaPietroTempZ (Device=plc1212 MemAddress=DB19.DBD374/DB19.DBD374 DataType=FLOAT Direction=0)
> 12:47:32.063 [DEBUG] [o.o.b.s.i.SimaticGenericDevice:550  ] - isoTCP:192.168.254.99 - Incoming data - item:PLC_LazienkaPietroTempZ/state:0

grzegorz914, 2018-01-19:
Hi,

just please remove "O" from all PLC adres or use "IO" and then test again.
With read (incoming data) is all OK, problem is if I want to write data to PLC above address 304.

thanks

docbender, 2018-01-19:
I had taken "O".
Write data give me error. Data are written into PLC but PLC return unexpected message. This message is unknown for used libnodave library. To fix this PLC answer must be implemented into this library

grzegorz914, 2018-01-19:
If I look also on the error from PLC, write data areas error is( DBB208-DBB374) however this is not true because I use DBD data (Real).

>2018-01-18 11:23:08.172 [ERROR] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - Read data area error (Area=DB19.DBB208-DB19.DBB374, Return code=0xffffff80, Error=Unexpected function code in answer)

grzegorz914, 2018-01-30:
Hi,
I have again strange error with connection to s7, any help?

`2018-01-30 05:50:16.202 [ERROR] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - Read data area error (Area=DB24.DBB0-DB24.DBB4, Return code=0xffffff7e, Error=Buffer size insufficient while readBytes)`

docbender, 2018-02-01:
Hi @grzegorz914,
can you test this build please: https://github.com/docbender/openHAB-Simatic/releases/download/1.1.3/org.openhab.binding.simatic-1.9.0-SNAPSHOT.jar
I looked at library socket operations and did some improvements. It looks better for me now.

grzegorz914, 2018-02-01:
Hi @docbender ,
many thanks for the update, with this release no any error witch communication look like all is OK, one other problem some data from DB6 is not read, nothing in openHAB log, if value changed on PLC site.

This is read correct and data is changed in openHAB log:
`
Switch    PLC_TermostatyTrybRecznyAuto   "Tryb pracy ogrzewania"  <switch >  (gTermostaty) { simatic="plc:DB6.DBX12.0:byte" }
2018-02-01 16:14:55.020 [vent.ItemStateChangedEvent] - PLC_TermostatyTrybRecznyAuto changed from ON to OFF
2018-02-01 16:15:28.691 [vent.ItemStateChangedEvent] - PLC_TermostatyTrybRecznyAuto changed from OFF to ON`

This all are not read, if value on PLC site changed then nothing changed in openHAB log.
`
Switch    PLC_TermostatyTrybRecznyAuto   "Tryb pracy ogrzewania"  <switch >  (gTermostaty) { simatic="plc:DB6.DBX12.0:byte" }

Number    PLC_PrzedpokojMalyTermostatStan    "Przedpokój mały"  <radiator> { simatic="plc:DB6.DBB23:byte" }
Number    PLC_PrzedpokojDuzyTermostatStan    "Przedpokój duży"  <radiator> { simatic="plc:DB6.DBB37:byte" }
Number    PLC_PokojParterTermostatStan    "Pokój parter"  <radiator> { simatic="plc:DB6.DBB51:byte" }
Number    PLC_SalonTermostatStan    "Salon"  <light> { simatic="plc:DB6.DBB65:byte" }
Number    PLC_JadalniaTermostatStan    "Jadalnia"  <radiator> { simatic="plc:DB6.DBB79:byte" }
Number    PLC_KuchniaTermostatStan    "Kuchnia"  <radiator> { simatic="plc:DB6.DBB93:byte" }
Number    PLC_LazienkaParterTermostatStan    "Łazienka parter"  <radiator> { simatic="plc:DB6.DBB107:byte" }
Number    PLC_GarderobaTermostatStan    "Garderoba"  <radiator> { simatic="plc:DB6.DBB121:byte" }
Number    PLC_BiuroTermostatStan    "Biuro"  <radiator> { simatic="plc:DB6.DBB135:byte" }
Number    PLC_KamilaTermostatStan    "Kamila"  <radiator> { simatic="plc:DB6.DBB149:byte" }
Number    PLC_SypialniaTermostatStan    "Sypialnia"  <radiator> { simatic="plc:DB6.DBB163:byte" }
Number    PLC_LazienkaPietroTermostatStan    "Łazienka piętro"  <radiator> { simatic="plc:DB6.DBB177:byte" }`

very strange.

All other DB and I,Q,M area are read correct.

docbender, 2018-02-02:
Can you collect trace log data from simatic binding? Maybe there will be some usefull information.

grzegorz914, 2018-02-02:
Here file from wireshark:

[simatic.zip](https://github.com/docbender/openHAB-Simatic/files/1688874/simatic.zip)

docbender, 2018-02-02:
Thanks for traffic log. I'm interested also in how binding reacts to your configuration. If db6 is in readareas and if it reads db6 and read data are resolved
To enable trace log type
 

> log:set TRACE org.openhab.binding.simatic

grzegorz914, 2018-02-02:
`
30 - PostValue for item=PLC_SypialniaTempZ (Device=plc MemAddress=DB6.DBD164/DB6.DBD164 DataType=FLOAT Direction=0)
2018-02-02 12:32:52.321 [TRACE] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - PostValue for item=PLC_LazienkaPietroTermostatStan (Device=plc MemAddress=DB6.DBB177/DB6.DBB177 DataType=BYTE Direction=0)
2018-02-02 12:32:52.323 [TRACE] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - PostValue for item=PLC_LazienkaPietroTempZ (Device=plc MemAddress=DB6.DBD178/DB6.DBD178 DataType=FLOAT Direction=0)
2018-02-02 12:32:55.560 [DEBUG] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - Reading finished. Area=DB6.DBB12-DB6.DBB182
2018-02-02 12:32:55.563 [TRACE] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - PostValue for item=PLC_TermostatyTrybRecznyAuto (Device=plc MemAddress=DB6.DBX12.0/DB6.DBX12.0 DataType=BYTE Direction=0)
2018-02-02 12:32:55.568 [TRACE] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - PostValue for item=PLC_PrzedpokojMalyTermostatStan (Device=plc MemAddress=DB6.DBB23/DB6.DBB23 DataType=BYTE Direction=0)
2018-02-02 12:32:55.574 [TRACE] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - PostValue for item=PLC_PrzedpokojMalyTempZ (Device=plc MemAddress=DB6.DBD24/DB6.DBD24 DataType=FLOAT Direction=0)
2018-02-02 12:32:55.580 [TRACE] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - PostValue for item=PLC_PrzedpokojDuzyTermostatStan (Device=plc MemAddress=DB6.DBB37/DB6.DBB37 DataType=BYTE Direction=0)
2018-02-02 12:32:55.585 [TRACE] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - PostValue for item=PLC_PrzedpokojDuzyTempZ (Device=plc MemAddress=DB6.DBD38/DB6.DBD38 DataType=FLOAT Direction=0)
2018-02-02 12:32:55.591 [TRACE] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - PostValue for item=PLC_PokojParterTermostatStan (Device=plc MemAddress=DB6.DBB51/DB6.DBB51 DataType=BYTE Direction=0)
2018-02-02 12:32:55.597 [TRACE] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - PostValue for item=PLC_PokojParterTempZ (Device=plc MemAddress=DB6.DBD52/DB6.DBD52 DataType=FLOAT Direction=0)
2018-02-02 12:32:55.602 [TRACE] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - PostValue for item=PLC_SalonTermostatStan (Device=plc MemAddress=DB6.DBB65/DB6.DBB65 DataType=BYTE Direction=0)
2018-02-02 12:32:55.607 [TRACE] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - PostValue for item=PLC_SalonTempZ (Device=plc MemAddress=DB6.DBD66/DB6.DBD66 DataType=FLOAT Direction=0)
2018-02-02 12:32:55.614 [TRACE] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - PostValue for item=PLC_JadalniaTermostatStan (Device=plc MemAddress=DB6.DBB79/DB6.DBB79 DataType=BYTE Direction=0)
2018-02-02 12:32:55.621 [TRACE] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - PostValue for item=PLC_JadalniaTempZ (Device=plc MemAddress=DB6.DBD80/DB6.DBD80 DataType=FLOAT Direction=0)
2018-02-02 12:32:55.628 [TRACE] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - PostValue for item=PLC_KuchniaTermostatStan (Device=plc MemAddress=DB6.DBB93/DB6.DBB93 DataType=BYTE Direction=0)
2018-02-02 12:32:55.636 [TRACE] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - PostValue for item=PLC_KuchniaTempZ (Device=plc MemAddress=DB6.DBD94/DB6.DBD94 DataType=FLOAT Direction=0)
2018-02-02 12:32:55.643 [TRACE] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - PostValue for item=PLC_LazienkaParterTermostatStan (Device=plc MemAddress=DB6.DBB107/DB6.DBB107 DataType=BYTE Direction=0)
2018-02-02 12:32:55.650 [TRACE] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - PostValue for item=PLC_LazienkaParterTempZ (Device=plc MemAddress=DB6.DBD108/DB6.DBD108 DataType=FLOAT Direction=0)
2018-02-02 12:32:55.656 [TRACE] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - PostValue for item=PLC_GarderobaTermostatStan (Device=plc MemAddress=DB6.DBB121/DB6.DBB121 DataType=BYTE Direction=0)
2018-02-02 12:32:55.662 [TRACE] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - PostValue for item=PLC_GarderobaTempZ (Device=plc MemAddress=DB6.DBD122/DB6.DBD122 DataType=FLOAT Direction=0)
2018-02-02 12:32:55.671 [TRACE] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - PostValue for item=PLC_BiuroTermostatStan (Device=plc MemAddress=DB6.DBB135/DB6.DBB135 DataType=BYTE Direction=0)
2018-02-02 12:32:55.678 [TRACE] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - PostValue for item=PLC_BiuroTempZ (Device=plc MemAddress=DB6.DBD136/DB6.DBD136 DataType=FLOAT Direction=0)
2018-02-02 12:32:55.684 [TRACE] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - PostValue for item=PLC_KamilaTermostatStan (Device=plc MemAddress=DB6.DBB149/DB6.DBB149 DataType=BYTE Direction=0)
2018-02-02 12:32:55.693 [TRACE] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - PostValue for item=PLC_KamilaTempZ (Device=plc MemAddress=DB6.DBD150/DB6.DBD150 DataType=FLOAT Direction=0)
2018-02-02 12:32:55.701 [TRACE] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - PostValue for item=PLC_SypialniaTermostatStan (Device=plc MemAddress=DB6.DBB163/DB6.DBB163 DataType=BYTE Direction=0)
2018-02-02 12:32:55.708 [TRACE] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - PostValue for item=PLC_SypialniaTempZ (Device=plc MemAddress=DB6.DBD164/DB6.DBD164 DataType=FLOAT Direction=0)
2018-02-02 12:32:55.714 [TRACE] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.1.30 - PostValue for item=PLC_LazienkaPietroTermostatStan (Device=plc MemAddress=DB6.DBB177/DB6.DBB177 Da
`

grzegorz914, 2018-02-02:
No problem, simatic.items

- [ //Pogoda

Number    PLC_PogodaTemp      "Temperatura powietrza"  <temperature>  [ "CurrentTemperature" ]   { simatic="plc:DB28.DBD190:float" }
Number    PLC_PogodaWilgotnosc      "Wilgotność powietrza"  <humidity>  [ "CurrentHumidity" ]   { simatic="plc:DB28.DBD202:float" }
Number    PLC_PogodaPunktRosy      "Punkt rosy"  <temperature>  { simatic="plc:DB28.DBD198:float" }
Number    PLC_PogodaWiatr               "Wiatr"  <wind>  { simatic="plc:DB28.DBD174:float" }
Number    PLC_PogodaOpady               "Opady"  <rain>  { simatic="plc:DB28.DBX242.0:byte" }
Number    PLC_PogodaCisnienie               "Ciśnienie atmosferyczne"  <pressure>  { simatic="plc:DB28.DBD210:float" }
Number    PLC_PogodaCzujnikZmierzchu        "Czujnik zmierzchu"  <text>  { simatic="plc:DB28.DBD238:DWORD" }
Number    PLC_PogodaEnergiaSlonca           "Energia słońca"  <text>  { simatic="plc:DB28.DBD218:float" }
Number    PLC_PogodaDlugoscGeograficzna    "Długość geograficzna"  <text>  { simatic="plc:DB28.DBD256:float" }
Number    PLC_PogodaSzerokoscGeograficzna    "Szerokość geograficzna"  <text>  { simatic="plc:DB28.DBD260:float" }
Number    PLC_PogodaWysokoscGeograficzna    "Wysokość geograficzna"  <text>  { simatic="plc:DB28.DBD272:DWORD" }
Number    PLC_PogodaSlonceElewacja    "Słońce elewacja"  <text>  { simatic="plc:DB28.DBD264:float" }
Number    PLC_PogodaSlonceAzymut    "Słońce azymut"  <text>  { simatic="plc:DB28.DBD268:float" }

//Wjazd

Switch    PLC_BramaTaster   "Brama"    <garagedoor>     ["Switchable"]    { simatic="plc:M2.0:byte" }
Switch    PLC_BramkaTaster   "Bramka"    <garagedoor>     ["Switchable"]    { simatic="plc:M2.1:byte" }

//Tastery_i_oswietlenie

Switch    PLC_PrzedpokojMalyTaster   "Przedpokój mały"    <switch>     ["Switchable"]    {simatic="plc:M20.0:byte"}
Switch    PLC_PrzedpokojDuzyTaster    "Przedpokój duży"    <switch>     ["Switchable"]    { simatic="plc:M20.1:byte" }
Switch    PLC_KotlowniaTaster   "Kotłownia"    <switch>     ["Switchable"]    { simatic="plc:M20.2:byte" }
Switch    PLC_PrzedsionekKotlowniTaster    "Przedsionek kotłowni"    <switch>     ["Switchable"]    { simatic="plc:M20.3:byte" }
Switch    PLC_PokojSufit1Taster   "Pokój sufit 1"    <switch>     ["Switchable"]    { simatic="plc:M21.5:byte" }
Switch    PLC_PokojSufit2Taster    "Pokój sufit 2"    <switch>     ["Switchable"]    { simatic="plc:M21.6:byte" }
Switch    PLC_SpizarniaTaster   "Spiżarnia"    <switch>     ["Switchable"]    { simatic="plc:M21.4:byte" }
Switch    PLC_SalonSufitTaster    "Salon sufit"    <switch>     ["Switchable"]    { simatic="plc:M20.4:byte" }
Dimmer    PLC_SalonSufitDimmer   "Salon sufit"    <dimmer>     ["Switchable"]    { simatic="plc:DB26.DBB50:byte" }
Switch    PLC_SalonKinkietyTaster    "Salon kinkiety"    <switch>     ["Switchable"]    { simatic="plc:M20.6:byte" }
Dimmer    PLC_SalonKinkietyDimmer   "Salon kinkiety"    <dimmer>     ["Switchable"]    { simatic="plc:DB26.DBB49:byte" }
Switch    PLC_JadalniaKinkietyTaster    "Jadalnia kinkiety"    <switch>     ["Switchable"]    { simatic="plc:M20.7:byte" }
Dimmer    PLC_JadalniaKinkietyDimmer   "Jadalnia kinkiety"    <dimmer>     ["Lighting"]    { simatic="plc:DB26.DBB48:byte" }
Switch    PLC_KuchniaSufitTaster    "Kuchnia sufit"    <switch>     ["Switchable"]    { simatic="plc:M21.0:byte" }
Dimmer    PLC_KuchniaSufitDimmer   "Kuchnia sufit"    <dimmer>     ["Switchable"]    { simatic="plc:DB26.DBB47:byte" }
Switch    PLC_KuchniaBlatTaster    "Kuchnia blat"    <switch>     ["Switchable"]    { simatic="plc:M21.1:byte" }
Dimmer    PLC_KuchniaBlatDimmer   "Kuchnia blat"    <dimmer>     ["Switchable"]    { simatic="plc:DB26.DBB44:byte" }
Switch    PLC_LazienkaParterSufitTaster    "Łazienka parter sufit"    <switch>     ["Switchable"]    { simatic="plc:M21.2:byte" }
Switch    PLC_LazienkaParterLustroTaster    "Łazienka parter lustro"    <switch>     ["Switchable"]    { simatic="plc:M21.3:byte" }

Switch    PLC_StrychTaster   "Strych"    <switch>     ["Switchable"]    { simatic="plc:M24.0:byte" }
Switch    PLC_LazienkaPietroSufitTaster   "Łazienka piętro sufit"    <switch>     ["Switchable"]    { simatic="plc:M24.1:byte" }
Switch    PLC_LazienkaPietroLustroTaster    "Łazienka piętro lustro"    <switch>     ["Switchable"]    { simatic="plc:M24.2:byte" }
Switch    PLC_LazienkaPietroLedTaster    "Łazienka piętro LED"    <switch>     ["Switchable"]    { simatic="plc:M25.5:byte" }
Switch    PLC_GarderobaSufit1Taster    "Garderoba sufit 1"    <switch>     ["Switchable"]    { simatic="plc:M24.3:byte" }
Switch    PLC_GarderobaSufit2Taster    "Garderoba sufit 2"    <switch>     ["Switchable"]    { simatic="plc:M24.4:byte" }
Switch    PLC_BiuroSufit1Taster    "Biuro sufit 1"    <switch>     ["Switchable"]    { simatic="plc:M24.5:byte" }
Switch    PLC_BiuroSufit2Taster    "Biuro sufit 2"    <switch>     ["Switchable"]    { simatic="plc:M24.6:byte" }
Switch    PLC_KamilaSufit1Taster    "Kamila sufit 1"    <switch>     ["Switchable"]    { simatic="plc:M24.7:byte" }
Switch    PLC_KamilaSufit2Taster    "Kamila sufit 2"    <switch>     ["Switchable"]    { simatic="plc:M25.0:byte" }
Switch    PLC_SypialniaKinkiet1Taster    "Sypialnia kinkiet 1"    <switch>     ["Switchable"]    { simatic="plc:M25.1:byte" }
Switch    PLC_SypialniaKinkiet2Taster    "Sypialnia kinkiet 2"    <switch>     ["Switchable"]    { simatic="plc:M25.2:byte" }
[... cut ...]
Number    PLC_PowAlarmNapad  "Alarm napadowy"    <text>   { simatic="plc:DB24.DBX2.5:byte" }
Number    PLC_PowPiecWylaczony  "Wyłączenie pieca"    <text>   { simatic="plc:DB24.DBX2.6:byte" }
Number    PLC_PowAlarmTrybNormalny "Powrót alarmu do trybu normalnego"    <text>   { simatic="plc:DB24.DBX2.7:byte" }
Number    PLC_PowBramkaZamknieta   "Zamknięcie bramki"    <text>   { simatic="plc:DB24.DBX3.0:byte" }
Number    PLC_PowBramaZamknieta   "Zamknięcie bramy"    <text>   { simatic="plc:DB24.DBX3.1:byte" }

]

grzegorz914, 2018-02-03:
Here full somatic log file:
[openhab.log.6.zip](https://github.com/docbender/openHAB-Simatic/files/1691444/openhab.log.6.zip)

docbender, 2018-02-05:
I added new release https://github.com/docbender/openHAB-Simatic/releases/tag/1.1.4
Please test it. I tested it with your configuration and DB6 is now read correctly on my system

grzegorz914, 2018-02-05:
Hello,

thanks for Your god work, all is OK now.

docbender, 2018-02-09:
Thanks for your testing.

closed 2018-02-09 by docbender