docbender/openHAB-Simatic #4: Binding stop working if Mail Action remove

grzegorz914, 2018-02-19:
Hello,

I have strange problem, after remove Mail Action from openHAB simatic bindings stop working with error:

> 2018-02-19 16:22:54.128 [ERROR] [org.openhab.binding.simatic         ] - FrameworkEvent ERROR - org.openhab.binding.simatic
> org.osgi.framework.BundleException: Could not resolve module: org.openhab.binding.simatic [222]
>   Unresolved requirement: Import-Package: org.openhab.core.binding
> 	at org.eclipse.osgi.container.Module.start(Module.java:444) [?:?]
> 	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1620) [?:?]
> 	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1600) [?:?]
> 	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.doContainerStartLevel(ModuleContainer.java:1571) [?:?]
> 	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1514) [?:?]
> 	at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1) [?:?]
> 	at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:230) [?:?]
> 	at org.eclipse.osgi.framework.eventmgr.EventManager$EventThread.run(EventManager.java:340) [?:?]

after install Mail Action again ,somatic start working correct.

docbender, 2018-02-20:
For me it looks like _Mail Action_ issue or even OH2 issue. When u uninstall it, org.openhab.core.binding dependency package is missing. It can not happen if there exists a package which need it (which is SIMATIC)

grzegorz914, 2018-03-07:
Ok,

I have other read error DB11 nad DB12 on other S7-1200 1214C, but DB15 is read correct.

> 2018-03-07 18:49:01.227 [ERROR] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.2.50 - Read data area error (Area=DB11.DBB0-DB11.DBB10, Return code=0x5, Error=the desired address is beyond limit for this PLC)
> 2018-03-07 18:49:01.480 [ERROR] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.2.50 - Read data area error (Area=DB12.DBB0-DB12.DBB10, Return code=0x5, Error=the desired address is beyond limit for this PLC)

Thanks

closed 2018-03-19 by docbender