docbender/openHAB-Simatic #5: how to install on openhab 2.2

artexx, 2018-03-27:
sorry for use of issue, I'm new to openhab and interested to connect it to my home plc.
I don't uderstant from you wiki how to add the config file since they are not used anymore from openhab.
Is it possible to use your addon with openhab 2.2?
Thanks in advance
Walter

docbender, 2018-03-27:
Hi,
you can run this binding under OH2 in compatibility mode: http://docs.openhab.org/developers/development/compatibilitylayer.html#how-to-use-openhab-1x-add-ons-that-are-not-part-of-the-distribution

Here is the part that should interest you:

While the openHAB distribution already contains many add-ons of openHAB 1, there are still quite some of them missing - please help testing them - if they are confirmed to be working, they can be included in the distribution. Test a not included add-on is very straight forward:
   1. Start your runtime
   2. Install the 1.x compatibility layer by running feature:install openhab-runtime-compat1x in the openHAB console
   3. As with openHAB 1.x, simply take the jar file of your add-on(org.openhab.binding.simatic-1.9.0-SNAPSHOT.jar) and place it in the ${openhab.home}/addons folder.
   4. Copy your personal openhab.cfg file to ${openhab.home}/conf/services/openhab.cfg

artexx, 2018-03-28:
thank you very much for your quick reply,
i have followed your instructions and run "feature:install openhab-runtime-compat1x" in console withour error message.
created a openhab.cfg in /conf/services
with few instruction just for testing

################################### Simatic Binding ######################################
#

 simatic:refresh=2000      
 simatic:plc=192.168.0.177:0.2
 Number    PortState            "PLC state [%s]"                 { simatic="plc:info:state" }
 Number    PortPreviouState     "PLC previous state [%s]"        { simatic="plc:info:previous_state" }
 DateTime  PortStateChangeTime  "PLC changed [%1$tA, %1$td.%1$tm.%1$tY %1$tT]"    { simatic="plc:info:state_change_time" }.

Restarted openhab
but I don't find any reference or message related to simatic bind 
can you please address me on how to check is binding has been included?
Thanks again
Walter

docbender, 2018-03-30:
You can check log by _log:list_ command.
Also you can enable debug: 
_log:set DEBUG org.openhab.binding.simatic_

artexx, 2018-03-31:
Thank you again for you r support.
I found on the log that the connection is open.
Now I'm starting to learn how to create items and how to use them.
For single bit merker the data type is "byte" if I understand correctly your wiki , can you please confirm?
Many thanks for your great work
Walter

docbender, 2018-04-01:
Yes. There is no example with bit address but you are right.  It is represent in OH as byte number with values 0 and 1

docbender, 2018-04-17:
What i know IOT2000 should be free programmable piece of hardware. So it depends on you what kind of protocol you use to connect to openHAB. For Simatic S7 protocol (this binding use it) I found this example https://support.industry.siemens.com/tf/ww/en/posts/s7-communication-with-node-red/159131/?page=0&pageSize=10

sephiroth6, 2018-04-17:
thanks for reply! The devices dialogue with MQTT.
With node-red, installed on another server with docker image, everything works. Now I would use this smart gateway IoT2040 and try it with openHAB. Searching, I found your git-repo :)

grzegorz914, 2019-01-15:
Hi

any way to port This binding to OH2.x, I use this successfully in 1.x compatibility mode but every time if upgrade OH need to enable this manually. Also S7 connection state do not work correct in OH2.x.?

Thanks
Grzegorz

docbender, 2019-01-27:
I'm sorry, but I do not plan to move to OH2 right now. There is no free time and there is no hunt after this also.

closed 2020-09-17 by docbender