docbender/openHAB-Simatic #7: Read data aera erroe

Chicken123456, 2019-01-27:
Hi,

I use Openhab 2.4 with the simatic binding 1.9.0. It runs stable and I haven´t any problems till I add an new variable. If I refresh the items with the added variable it runs in an circle:

2019-01-27 17:49:40.143 [INFO ] [el.core.internal.ModelRepositoryImpl] - Refreshing model 'plc.items'
2019-01-27 17:49:44.302 [ERROR] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.2.50 - Read data area error (Area=I8-I26, Return code=0xffffff7e, Error=Buffer size insufficient while readBytes)
==> /var/log/openhab2/events.log <==
2019-01-27 17:49:44.454 [vent.ItemStateChangedEvent] - PortState changed from 1 to 6
==> /var/log/openhab2/openhab.log <==
2019-01-27 17:49:46.311 [INFO ] [imatic.internal.SimaticGenericDevice] - isoTCP:192.168.2.50: Trying to reconnect
==> /var/log/openhab2/events.log <==
2019-01-27 17:49:46.323 [vent.ItemStateChangedEvent] - PortState changed from 6 to 2
==> /var/log/openhab2/openhab.log <==
2019-01-27 17:49:46.822 [INFO ] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.2.50 -
connected
==> /var/log/openhab2/events.log <==
2019-01-27 17:49:46.831 [vent.ItemStateChangedEvent] - PortState changed from 2 to 1
==> /var/log/openhab2/openhab.log <==
2019-01-27 17:49:47.869 [ERROR] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.2.50 - Read data area error (Area=I8-I26, Return code=0xffffff7e, Error=Buffer size insufficient while readBytes)
==> /var/log/openhab2/events.log <==
2019-01-27 17:49:47.936 [vent.ItemStateChangedEvent] - PortState changed from 1 to 6
==> /var/log/openhab2/openhab.log <==
2019-01-27 17:49:49.873 [INFO ] [imatic.internal.SimaticGenericDevice] - isoTCP:192.168.2.50: Trying to reconnect
==> /var/log/openhab2/events.log <==
2019-01-27 17:49:49.883 [vent.ItemStateChangedEvent] - PortState changed from 6 to 2
==> /var/log/openhab2/openhab.log <==
2019-01-27 17:49:50.382 [INFO ] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.2.50 - connected
==> /var/log/openhab2/events.log <==
2019-01-27 17:49:50.391 [vent.ItemStateChangedEvent] - PortState changed from 2 to 1
==> /var/log/openhab2/openhab.log <==
2019-01-27 17:49:51.422 [ERROR] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.2.50 - Read data area error (Area=I8-I26, Return code=0xffffff7e, Error=Buffer size insufficient while readBytes)
==> /var/log/openhab2/events.log <==
2019-01-27 17:49:51.531 [vent.ItemStateChangedEvent] - PortState changed from 1 to 6
==> /var/log/openhab2/openhab.log <==
2019-01-27 17:49:53.426 [INFO ] [imatic.internal.SimaticGenericDevice] - isoTCP:192.168.2.50: Trying to reconnect
==> /var/log/openhab2/events.log <==
2019-01-27 17:49:53.442 [vent.ItemStateChangedEvent] - PortState changed from 6 to 2
==> /var/log/openhab2/openhab.log <==
2019-01-27 17:49:53.941 [INFO ] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.2.50 - connected
==> /var/log/openhab2/events.log <==
2019-01-27 17:49:53.947 [vent.ItemStateChangedEvent] - PortState changed from 2 to 1

**my simatic.cfg:**

# Konfigurations Datei fÃ¼r Simatic binding

# refresh-time in ms
simatic:refresh=2000
simatic:plc=192.168.2.50:0.1:S7

The problem exists only if I add an new inputvariable. I think if I add more outputvariable, it will take the same problem with the writebuffer. I have round about 200 variables and a lot of them are MD's.

Thanks!

docbender, 2019-01-27:
Hi, all binding version are 1.9.0 because it is OH version for which is it compiled. Right binding version you can see in DEBUG mode. It should be org.openhab.binding.simatic_1.9.0.201802051023.
To your issue. Yes from log i can see it is Input area problem I8-I26. Please post/attach your items configuration so I can simulate issue.

Chicken123456, 2019-01-27:
Hi,

I changed the binding to the newest version and the problem is solved. I downloaded the old version in august or september from github.

Thanks!

closed 2019-01-28 by docbender