docbender/openHAB-Simatic #8: Slow response and system faulty

grzegorz914, 2019-05-20:
Hi,

I have regularly updated OH2 to latest snapshot now I’m on 2.5.0-1996, from less than two months simatic binding is a terrible slower complete system and consequently leads to a total system failure. I thing this behavior corespondent with latest changes in OH2. Downgrade to 2.5.0-M1 solves this problem.

Can U check this behavior and fixed if possible?

Thank 
Grzegorz

docbender, 2019-05-21:
This has nothing with addon. This is OH problem. https://community.openhab.org/t/openhab2-snapshot-2-5-1595-fully-broken/74702/9

⁣Odesláno z TypeApp ​

grzegorz914, 2019-05-21:
Strange, because all other installed bindings working OK if Simatic.binding is not installed.

grzegorz914, 2019-05-26:
Hi,

I make some test, 
1. installed OH 2.5.0 build 1601 from scratch, installed other binding without somatic binding CPU Usage 5-10%. Installed simatic - CPU Usage over 200%, system slow down.
```
  PID     USER      PR    NI      VIRT      RES         SHR S        %CPU      %MEM            TIME+        COMMAND                                                                                                                                                              
 2751  openhab   20    0   627468 436880.    14792 S       212.8.         43.7           33:59.91           java                                                                                                                                                                 
 3088 openhab+  20   0    9892      3332           2768 R        1.0.            0.3             0:15.90             top                                                                                                                                                                  
 10.        root          20   0       0            0                 0 I              0.3             0.0             0:01.70 rcu_sched                                                                                                                                                            
  389 avahi            20   0    6528      2836           2536 S        0.3            0.3             0:04.00 avahi-daemon                                                                                                                                                         
 3470 root             20   0       0           0                0 I                0.3              0.0         0:00.16 kworker/2:0-eve                                                       
````
2. Installed OH2.5.0 build 1601 from scratch, installed only somatic without any other binding, CPU Usage over 200% as above.

any tips to find what is wrong?

thanks
Grzegorz

docbender, 2019-05-27:
> Komu: docbender/openHAB-Simatic <openHAB-Simatic@noreply.github.com>
> 8)
> "

Hi,

This must be something strange in OH latest snapshots. There are some 
changes around replacing ESH components. I recommend to stay at M1 version 
and wait to next milestone version. Or you can report this issue to OH 
developers (check this topic: https://github.com/openhab/openhab-core/
issues/829)

Best

 
 ---------- Původní e-mail ----------

   

<code>  PID     USER      PR    NI      VIRT      RES         SHR S        %CPU      %MEM            TIME+        COMMAND                                                                                                                                                              
</code>

   1. Installed OH2.5.0 build 1601 from scratch, installed only somatic 
   

(https://github.com/docbender/openHAB-Simatic/issues/8?email_source=notifications&email_token=ACPEJ3WC2K4EF5QBHMOIO4DPXK667A5CNFSM4HN666W2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODWIJSLI#issuecomment-496015661)
"

grzegorz914, 2019-05-28:
OK, 

already reported it.

grzegorz914, 2019-06-25:
Problem solved with 2.5.0 build 1619.

closed 2019-06-25 by grzegorz914