docbender/openHAB-Simatic #9: Error message in Openhab

Maddin2020, 2019-07-25:
Hi, first thank you so much for this binding. I use it a long time and now i want read some values more and i have a problem. In the log from OH 2 i see thome faults. Here a example

2019-07-26 01:21:51.066 [ERROR] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.178.200 - Read data area error (Area=DB2001.DBB1400-DB2001.DBB1425, Return code=0xffffff7d, Error=No data returned. Connection lost?)

Here my items from this error log

`// Leuchten
Switch Leuchte_B_Bad                                                                                                     {simatic="plc:DB2001.DBB1400"}
Switch Leuchte_ST_Bad                                                                                                          {simatic="plc:DB2001.DBB1401"}
Switch Leuchte_B_Bad_Dusche                                                                      {simatic="plc:DB2001.DBB1402"}
Switch Leuchte_ST_Bad_Dusche                                                                                                   {simatic="plc:DB2001.DBB1403"}
Switch Leuchte_B_Bad_Spiegel                                                                    {simatic="plc:DB2001.DBB1404"}
Switch Leuchte_ST_Bad_Spiegel                                                                                                  {simatic="plc:DB2001.DBB1405"}

// Kontakte
Number    Status_Fenster_Bad_Ost                                                                                           {simatic="plc:DB2001.DBX1406.0"}
Number    Status_Fenster_Bad_Nord                                                                  {simatic="plc:DB2001.DBX1406.1"}

//Jalousie Ost
Rollershutter Jalousie_Steuerung_Bad_Ost                                                            {simatic="plc:DB2001.DBW1408"}
Number        JalousieIstposition_Bad_Ost                                                          {simatic="plc:DB2001.DBW1410"}  
Number        JalousieSollposition_Bad_Ost                                                        {simatic="plc:DB2001.DBW1412"}
Switch        Setup_Bad_Ost_auto_close
{simatic="plc:DB2001.DBX1414.0"}
Switch        Setup_Bad_Ost_auto_open                                                 {simatic="plc:DB2001.DBX1414.1"}

//Jalousie Nord
Rollershutter Jalousie_Steuerung_Bad_Nord                                                     {simatic="plc:DB2001.DBW1418"}
Number        JalousieIstposition_Bad_Nord
{simatic="plc:DB2001.DBW1420"}  
Number        JalousieSollposition_Bad_Nord                                                  {simatic="plc:DB2001.DBW1422"}
Switch        Setup_Bad_Nord_auto_close                                                {simatic="plc:DB2001.DBX1424.0"}
Switch        Setup_Bad_Nord_auto_open                                                   {simatic="plc:DB2001.DBX1424.1"}`

And here my simatic.cfg for a S7 1500 CPU is the follow

simatic:refresh=200
simatic:plc=192.168.0.1:0.1

The DB and the values in the SPS are available.

Thanks for help

Maddin2020, 2019-07-29:
Here same details from the debug /trace log 
2019-07-29 01:58:54.328 [TRACE] [imatic.internal.SimaticGenericDevice] - isoTCP:192.168.178.200 - Incoming data - item:Steckdose_ST_Terrasse_Wand/state:OFF

2019-07-29 01:58:56.584 [ERROR] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.178.200 - Read data area error (Area=DB2001.DBB3000-DB2001.DBB3005, Return code=0xffffff7d, Error=No data returned. Connection lost?)

2019-07-29 01:58:56.586 [DEBUG] [.binding.simatic.internal.SimaticTCP] - Unlocking

2019-07-29 01:58:57.590 [DEBUG] [ding.simatic.internal.SimaticBinding] - execute() method is called!

2019-07-29 01:58:57.592 [DEBUG] [imatic.internal.SimaticGenericDevice] - reconnectWithDelaying(): 0/0/15

2019-07-29 01:58:57.594 [INFO ] [imatic.internal.SimaticGenericDevice] - isoTCP:192.168.178.200: Trying to reconnect

2019-07-29 01:58:57.596 [DEBUG] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.178.200 - close() - disconnecting

2019-07-29 01:58:57.598 [DEBUG] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.178.200 - open() - connecting

2019-07-29 01:58:59.115 [INFO ] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.178.200 - connected

2019-07-29 01:58:59.116 [DEBUG] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.178.200 - checkNewData() is called

2019-07-29 01:58:59.118 [DEBUG] [.binding.simatic.internal.SimaticTCP] - Locking

docbender, 2019-07-29:
Hi. 
At first check binding version. Right binding version you can see in DEBUG mode. It should be org.openhab.binding.simatic_1.9.0.201802051023.
Your problem looks like bad any pointer assembly, because you are using high DB addresses. I can check this at end of the week. If you need quick solution, I would recommend to you use lower address. Not DB2001.DBB3000 but something like DB2.DBB0

Maddin2020, 2019-08-02:
Hello again, thanks a lot for support. My version is the follow 1.9.0.201901290812. I think its a newer version, or not. I will check to use lower DB adresses. What i don´t understand is that i don´t have problems with openhab because the communication reconnect again and the data will transfer, i have only the errors in the log. It´s ok to receive the values every 200 ms or is that wrong. I have change the cycle from 200 ms to 1000 ms but no change. 

I hope you find the fault. If you need same data from me let me know it. 

Thanks again, Greetz martin

docbender, 2019-08-02:
Sorry I did not realise that there is newer version :-)
If data are read without problem so it should not be addressing problem. I'll test it and we will see

docbender, 2019-08-02:
I just tested configuration you posted with my PLC and it runs without any problem for hour. Even with 200ms refresh. 
Could it not be a high network response problem? Used library expected fast response from PLC. Read timeout is around 250ms.

closed 2020-09-17 by docbender