docbender/openHAB-Simatic #10: Response of values is very slow

th3elk, 2019-11-24:
Hi,
first I want to thank you for the great binding. I use openHAB 2.4 and binding version 1.9.0.201901290812 with a PLC 1513F.
I use about 240 Items at the moment. If I send a command to the PLC, for example I switch on a light by a dimmer, it reacts always immediately. But the status of the Item takes between 1 and 15 seconds to update, as you can see in the eventlog:

2019-11-24 18:19:59.057 [ome.event.ItemCommandEvent] - Item 'OG_Li_Wohnen_Esstisch' received command 0
2019-11-24 18:20:06.813 [vent.ItemStateChangedEvent] - OG_Li_Wohnen_Esstisch changed from 63 to 0
2019-11-24 18:22:42.397 [ome.event.ItemCommandEvent] - Item 'OG_Li_Wohnen_Esstisch' received command 59
2019-11-24 18:22:52.607 [vent.ItemStateChangedEvent] - OG_Li_Wohnen_Esstisch changed from 0 to 59
2019-11-24 18:24:53.049 [ome.event.ItemCommandEvent] - Item 'OG_Li_Wohnen_Esstisch' received command 0
2019-11-24 18:25:04.919 [vent.ItemStateChangedEvent] - OG_Li_Wohnen_Esstisch changed from 59 to 0
2019-11-24 18:25:36.890 [ome.event.ItemCommandEvent] - Item 'OG_Li_Wohnen_Esstisch' received command 47
2019-11-24 18:25:38.488 [vent.ItemStateChangedEvent] - OG_Li_Wohnen_Esstisch changed from 0 to 47
2019-11-24 18:25:40.049 [ome.event.ItemCommandEvent] - Item 'OG_Li_Wohnen_Esstisch' received command 0
2019-11-24 18:25:55.355 [vent.ItemStateChangedEvent] - OG_Li_Wohnen_Esstisch changed from 47 to 0

The refresh-time is set to 200ms. But the behavior doesn’t chance if I change the refresh-rime to 20ms or 5000ms. There are no errors in the log-file.
Is that behavior normally, or is there a way to make the response quicker? Or is there any error in my configuration?

My binding-configuration looks like this:
simatic:refresh=200 
simatic:plc=192.168.178.54:0.1

And my Items I attached.

Thanks in advance
th3elk

[items.txt](https://github.com/docbender/openHAB-Simatic/files/3883954/items.txt)

grzegorz914, 2019-11-24:
Please try this config:

`simatic:plc=192.168.178.54:0.1:S7`

I use Simatic binding with my S71200 1215C read and write over 1000 items and response is very quickly.

th3elk, 2019-11-24:
Thanks for your quick reply. I tested your suggestion, but nothing changed. I also restarted the system.
If I reduce the number of Items, the response gets faster.

I attached the log of the binding:
[openhab.log](https://github.com/docbender/openHAB-Simatic/files/3883977/openhab.log)

grzegorz914, 2019-11-24:
Please also remove from all items Optional syntax I - input, O - output and IO and check again.

th3elk, 2019-11-24:
Thanks again. I removed all IO optional syntax, but nothing changed.
I took a closer look to the log of the binding. There I saw that every about 5ms an new value is read. But only if the Items are closed to each other in the DB on the PLC. If there is a space to the next items to read there is this entry:
[DEBUG] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.178.54 - Reading finished. Area=DB103.DBB1600-DB103.DBB1611

And then it takes about 200ms until the next value is read. I have a lot of this spaces between the used values to have some reserve in my DB.
Could this be the solution to delete the reserve? Or can the binding be forced to read the whole DB as one part?

grzegorz914, 2019-11-24:
I think this binding read only DB memory area which is defined in the items and not the whole DB, the reserve is just skipped. 
For detailed descriptions of this behavior U need wait on @docbender antwort. 
As I previous described I do not have any problems with time response.
I working on RPI4 with OH2.5 latest snapshot.

docbender, 2019-11-25:
Hi, 
if speed is at first place, you should optimise your transferred data. Ideal layout is put all data into one data area (datablock) one by one without any gaps. That's because data from PLC are read in blocks. One block can be up to 192 bytes. If data length is greater, another block read is established. This is driver depended. 
Similarly, there are a gaps. If gap between configured data if greater then 32 bytes, new block read is established. But this can be optimised by data layout. So gaps smaller than 32 bytes means less read from PLC.
As you can see from your debug log, binding has to read 46 data areas from your PLC:
> readAreas(Size=46)

> DB101.DBB0-DB101.DBB1
> DB101.DBB100-DB101.DBB101
> DB102.DBB0-DB102.DBB2
> DB102.DBB100-DB102.DBB103
> DB102.DBB150-DB102.DBB153
> DB102.DBB200-DB102.DBB203
> DB102.DBB250-DB102.DBB253
> DB102.DBB300-DB102.DBB303
> DB102.DBB350-DB102.DBB353
> DB102.DBB400-DB102.DBB403
> DB102.DBB450-DB102.DBB453
> DB102.DBB500-DB102.DBB503
> DB102.DBB550-DB102.DBB553
> DB103.DBB0-DB103.DBB2
> DB103.DBB100-DB103.DBB103
> DB103.DBB150-DB103.DBB153
> DB103.DBB200-DB103.DBB203
> DB103.DBB250-DB103.DBB253
> DB103.DBB300-DB103.DBB303
> DB103.DBB350-DB103.DBB353
> DB103.DBB400-DB103.DBB403
> DB103.DBB450-DB103.DBB453
> DB103.DBB500-DB103.DBB503
> DB103.DBB550-DB103.DBB553
> DB103.DBB800-DB103.DBB811
> DB103.DBB850-DB103.DBB861
> DB103.DBB900-DB103.DBB911
> DB103.DBB950-DB103.DBB961
> DB103.DBB1000-DB103.DBB1011
> DB103.DBB1050-DB103.DBB1061
> DB103.DBB1100-DB103.DBB1111
> DB103.DBB1150-DB103.DBB1161
> DB103.DBB1200-DB103.DBB1211
> DB103.DBB1250-DB103.DBB1261
> DB103.DBB1300-DB103.DBB1311
> DB103.DBB1350-DB103.DBB1361
> DB103.DBB1400-DB103.DBB1411
> DB103.DBB1450-DB103.DBB1461
> DB103.DBB1500-DB103.DBB1511
> DB103.DBB1550-DB103.DBB1561
> DB103.DBB1600-DB103.DBB1611
> DB103.DBB1650-DB103.DBB1661
> DB103.DBB1700-DB103.DBB1701
> DB104.DBB0-DB104.DBB1
> DB104.DBB100-DB104.DBB103
> DB104.DBB800-DB104.DBB801

Most of areas are only 1-3 bytes long. If you put all data together in row, it can be read in one or two cycles.

th3elk, 2019-11-26:
Hi,

thanks for your help! First I tried to add some filling items to get gaps smaller 32 byte. The status now gets updated every 5 seconds. So, your solution works. I will now go to reorganize DBs to get even faster.
What do you mean with the 192 byte maximum blocksize is driver depended? Is libnodave the driver? If it is so, could you tell me why 192 bytes? I read a little bit about libnodave and there was mentioned that the maximum blocksize depends on the PDU of the PLC. For my PLC the PDU is 960 byte. Do you think I could test  to increase the  “MAX_DATA_LENGTH = 192” in file SimaticReadWriteDataArea.java or is there another limitation?

During testing to speed up the communication I tried to split it on more connections. I added the PLC 4 times in the simatic.cfg. But it didn’t get faster. So I deleted the 3 added connections. But they still appeared in the log. Also a restart of the system didn’t delete them.
I found them also entered in the file userdata\config\org\openhab\simatic.config. When I deleted them there, they didn’t appear in the log anymore. Just for your info.

Regards
th3elk

docbender, 2019-11-26:
Maximum data length in one request is normally depends on PDU size (should be 240 byte for S7-300 and S7-1200, 480 bytes for S7-400, S7-1500 and WinAC). 
I'm not sure why I set it “MAX_DATA_LENGTH = 192”. It is few years ago. Probably some compatibility mode what I was tested as maximum possible for all PLC types. Nor I even expect that someone will use this for large project :-) So yes MAX_DATA_LENGTH is false limitation and it can be increased in some situation.

grzegorz914, 2019-11-26:
@docbender 
can You add "MAX_DATA_LENGTH =" to the .cfg file, it will be simple to modify them and test it.

Regards
Grzegorz

docbender, 2019-11-26:
Publishing this variable is not good idea. But depend by negotiated PDU size during connect I made the binding to use bigger data area. If negotiated PDU size is 480 bytes (libnodave probably did not ask for more), maximal data area in one request is now 462 bytes. If you are interest, you can find it here as [new release 1.2.1](https://github.com/docbender/openHAB-Simatic/releases/download/1.2.1/org.openhab.binding.simatic-1.14.0-SNAPSHOT.jar)

Maddin2020, 2019-11-26:
@docbender I have a question too. I have found a Information in the manual from the S7 1500 PLC. The maximum data length is 960. You can see it here. 
[image]

I have read too that if the length to long then libnodave will reduce it. If not then is it possible to create a special binding for S7 1500 users. 

Thank you a lot for create this binding.

th3elk, 2019-11-27:
@docbender thanks very much for your great work. I tested your new version of the binding and it works fine for me. Now I’m down to 1.3 seconds for refresh the items.

As @Maddin2020 mentioned it would be nice it would be nice to use the full power of the S7 1500 with 960 bytes. 
As I read the MAX_DATA_LENGTH should always be PDU-size minus 22 bytes header. I saw you get the PDU-size from libnodave. I think it would may be a good idea to calculate the MAX_DATA_LENGTH this way: MAX_DATA_LENGTH = pduSize – 22 So MAX_DATA_LENGTH would be flexible, even if there will be a more powerfull PLC sometime.

docbender, 2019-11-27:
Yeah, you right. It would be better if data limit per read request will be calculated. I just clarify that read data limit MAX_DATA_LENGTH = pduSize - 18 (overhead 14 bytes + 4 bytes per requested item) what gives 462 bytes for 480PDU. Similarly for write. That has overhead 28 bytes(12+16 per item).
About 960PDU. If I tested it, libnodave negotiated only 480PDU for a such PLC. Probably the library does not count on that.

th3elk, 2019-11-27:
Can I see somewhere waht is negotiated for my plc?

Maddin2020, 2019-11-27:
@docbender At the first thank you so much for create and change the binding. If it possible to change the binding to use optimize data blocks with symbol names. Then we can use symbol names in the items file. 

Do you have a S7 1500 ? I know that the PDU size was changed in the newer firmware.

grzegorz914, 2019-11-27:
Symbolic access will be very nice feature but as I know is not possible right now. 
The Siemens S7 protocol is not documented by Siemens.

docbender, 2019-11-27:
> Can I see somewhere waht is negotiated for my plc?

In log. Look for something like "PLC connected. PDU size = xxx
>If it possible to change the binding to use optimize data blocks with symbol names. Then we can use symbol names in the items file.

No. Symbol names are supported by S7Plus protocol. This binding knows only standard S7.
>I know that the PDU size was changed in the newer firmware

In which way size was changed? I thought 960 PDU has been there since ages.

th3elk, 2019-11-27:
I looked for my negotiated PDU size now and I got 960:

`2019-11-27 14:01:40.018 [DEBUG] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.178.54 - open() - connecting
2019-11-27 14:01:40.529 [INFO ] [.binding.simatic.internal.SimaticTCP] - isoTCP:192.168.178.54 - connected. Max PDU size = 960`

So for me it would be nice if the MAX_DATA_LENGTH could  be calculated.

docbender, 2019-11-28:
Here is [new release](https://github.com/docbender/openHAB-Simatic/releases/download/1.2.2/org.openhab.binding.simatic-1.14.0-SNAPSHOT.jar) with suggested data area calculation. Max read in one transaction are now PDU size - 18 bytes.

I have to correct my mistake about the PDU size negotiated by library with S7-1500. I said that maximum is only 480bytes. That was because I was connected through communication processor. Now I connect directly to the PLC and 960bytes PDU has been negotiated.

Maddin2020, 2019-11-28:
@docbender Thanks a lot !! GREAT JOB

grzegorz914, 2019-11-28:
@docbender 
Thanks, all working fine except showing connection state.
Right now my connection PortState is 2 - Closed and PreviousPortState is 1 - Listening, but this is not true because the connection still alive and data is read and write correct. 
This behavior was also before update the binding, just I report this right now.

Thanks
Grzegorz

th3elk, 2019-11-29:
@docbender 
Thanks a lot! The binding works great for me now!

th3elk, 2019-12-01:
Now I had a little time to test the new binding. I don’t get an update for some values. For me it looks like I don’t get an update for the items, that are read in one area above byte 490. The next read area is fine again. 
Can someone check this behavior for items above byte 490 in a read-area?
In the log, there is no entry. It looks like all items are updated correctly. But the Items above byte 490 always get the value 0.

docbender, 2020-01-14:
Today I looked at it. There was problem in libnodave, cos Simatic segmented data into two packets and libnodave processed only first one. Now it is fixed and all segmented PDU will be processed. 
So now all data from S7-1500 will be read (tested 900bytes data area).

You can test new release https://github.com/docbender/openHAB-Simatic/releases/download/1.2.3/org.openhab.binding.simatic-1.14.0-SNAPSHOT.jar

th3elk, 2020-01-30:
Thanks very much for your great work! I tested the new version of the binding and it works great for me now. I'm now down to 1.5s to read all values. And all values get updated now correctly.

Osmii, 2020-05-18:
Hello @th3elk.
Could you help me with rollershuters. I have Et200 Siemens, with 5 shutters. I have 2 buttons conected to PLC for Up and Down and two outputs for motor Up and motor Down.
I have no idea how to program fb with db to connect it with openhab. Could you provide me an example?
Thank you!

closed 2020-09-17 by docbender