docbender/openHAB-Simatic #12: removed 1.x compatibility layer #1284

grzegorz914, 2019-12-20:
Hi,

OH will removed 1.x compatibility layer #1284 in upcoming release, what is the future of the binding? will be adapted to the new system? or we need still in OH2.x version.

Thanks

docbender, 2019-12-21:
Yeah I know about this rumors. But I have not seen even OH3 roadmap. Main change should be remove eclipse smarthome dependency. It probably break even OH2 bindings compatibility. So until I see some OH3 beta (expecting in half a year), there is no reason for me to migrate binding. Then if I have spare time, I have plan to migrate

grzegorz914, 2019-12-21:
Ok,

thanks for clarifying that.

As long as Your binding will not be migrate to OH3 I will still on OH2.5.

closed 2019-12-21 by grzegorz914