docbender/openHAB-Simatic #14: Rollershutter item explaination.

Osmii, 2020-05-23:
Hello to all!
Could someboddy explain to me how the DB works with Rollershuter item.
If word has two bytes, 16 bits togheter and and first byte is for status Stop, Move,Up, Down second is for comannd 1-Move,2-Stop,4-Up,8-Down, that is 8 bits thogether.
If I understand corectly in first byte(status), bits are:
0-Stop
1-Move
3-Up
7-Down
2,4,5,6-are empty?

In second byte(command), bits are:
0-Move
1-Stop
3-Up
7-Down
2,4,5,6-are empty?

In sitemap commands are presented with ^ x ˇ simbols. How is status presented? With Stop, Up, etc. text?
If I create FB with for 4bits for status and 4 bits for commands in Static section of Block interface, how do I fill the gap between them?

Lets say that my first bit starts with address %DB27.DBX4.0 and my last ends with %DB27.DBX5.7.
What is exact definition of item, something like {simatic="plc:DB27.DBW4:word:IO"} ?

Thank you for your anwsers!

Maddin2020, 2020-05-23:
Hello i think i can help you First you can read the rolershutter in the Description 
In the Upper Byte you find the Position in the lower Byte you have the follow occupancy 
1-Move,2-Stop,4-Up,8-Down
If you want create a fb you Must Mask and demask the Bytes i you Need some Tips let my know it

Osmii, 2020-05-25:
So in uper byte position is presented with 0-100% ?
Does this use all 8 bits?
If you could give me an example for masking that would be great, becuse Im beginner in Siemens.

docbender, 2020-05-26:
You can test it easily. Assign your rollershutter item to address MW100 in PLC. And in PLC watch variables MB100 and MB101. What happens if you click or hold button on web in variable MB101? What if you set some value between 0-100 into MB100?

⁣Best

Osmii, 2020-05-28:
@docbender thans!
I tested it and if i press commands in OH, binary ouput in MB101 is 2 for Stop,4 for Up,8 for Down as stated. If i change value in MB100 the new value is visible in OH log.
I have two more questions:
- Why value is not visible in sitemap(only in log)?
- Why does value fall back to 0 when I press Up or Down command?

`2020-05-28 19:10:15.685 [vent.ItemStateChangedEvent] - DelavniskoOkno changed from 0 to 34

2020-05-28 19:11:00.917 [vent.ItemStateChangedEvent] - DelavniskoOkno changed from 34 to 100

2020-05-28 19:11:16.752 [ome.event.ItemCommandEvent] - Item 'DelavniskoOkno' received command UP

2020-05-28 19:11:18.522 [vent.ItemStateChangedEvent] - DelavniskoOkno changed from 100 to 0`

Thanks

docbender, 2020-06-01:
* Why it is not visible in sitemap I have no idea. OH issue or wrong sitemap configuration. 
* Value gets into 0 during command write. If you write status value every PLC scan, that will not be a problem.

Osmii, 2020-06-04:
@docbender  thanks!

closed 2020-06-04 by Osmii