docbender/openHAB-Simatic #18: BufferUnderflowException - Read data error OH3.0

grzegorz914, 2020-09-28:
Hi,

After create more as 152 Channels get communications error with PLC, 
only first 152 Channels are read correct all over not:

```
==> /var/log/openhab/events.log <==
2020-09-28 16:45:55.136 [INFO ] [marthome.event.ItemStateChangedEvent] - Item 'PLC_OczkoLampaUvCzasOdliczanie' changed from 2464731 to 2463959
2020-09-28 16:45:55.138 [INFO ] [marthome.event.ItemStateChangedEvent] - Item 'OczkoPompaCzasOdliczanieSec' changed from 2464.73100000 to 2463.95900000
2020-09-28 16:45:55.143 [INFO ] [marthome.event.ItemStateChangedEvent] - Item 'OczkoLampaUvCzasOdliczanieSec' changed from 2464.73100000 to 2463.95900000
2020-09-28 16:45:55.672 [INFO ] [marthome.event.ItemStateChangedEvent] - Item 'PLC_WodaCis' changed from 3.6024956703186035 to 3.6010489463806152

==> /var/log/openhab/openhab.log <==
2020-09-28 16:45:55.923 [ERROR] [nternal.simatic.SimaticGenericDevice] - 192.168.1.30 - Read data error
java.nio.BufferUnderflowException: null
        at java.nio.Buffer.nextGetIndex(Buffer.java:650) ~[?:?]
        at java.nio.HeapByteBuffer.getFloat(HeapByteBuffer.java:504) ~[?:?]
        at org.openhab.binding.simatic.internal.simatic.SimaticGenericDevice.postValue(SimaticGenericDevice.java:544) ~[bundleFile:?]
        at org.openhab.binding.simatic.internal.simatic.SimaticTCP.readDataArea(SimaticTCP.java:320) ~[bundleFile:?]
        at org.openhab.binding.simatic.internal.simatic.SimaticGenericDevice.checkNewData(SimaticGenericDevice.java:389) [bundleFile:?]
        at org.openhab.binding.simatic.internal.simatic.SimaticGenericDevice.execute(SimaticGenericDevice.java:136) [bundleFile:?]
        at org.openhab.binding.simatic.internal.simatic.SimaticGenericDevice.lambda$0(SimaticGenericDevice.java:109) [bundleFile:?]
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515) [?:?]
        at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:305) [?:?]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:305) [?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) [?:?]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) [?:?]
        at java.lang.Thread.run(Thread.java:834) [?:?]

==> /var/log/openhab/events.log <==
2020-09-28 16:45:55.926 [INFO ] [marthome.event.ItemStateChangedEvent] - Item 'PLC_OczkoPompaCzasOdliczanie' changed from 2463959 to 2463166
2020-09-28 16:45:55.930 [INFO ] [marthome.event.ItemStateChangedEvent] - Item 'PLC_OczkoLampaUvCzasOdliczanie' changed from 2463959 to 2463166
2020-09-28 16:45:55.931 [INFO ] [marthome.event.ItemStateChangedEvent] - Item 'OczkoPompaCzasOdliczanieSec' changed from 2463.95900000 to 2463.16600000
2020-09-28 16:45:55.935 [INFO ] [marthome.event.ItemStateChangedEvent] - Item 'OczkoLampaUvCzasOdliczanieSec' changed from 2463.95900000 to 2463.16600000
2020-09-28 16:45:56.458 [INFO ] [marthome.event.ItemStateChangedEvent] - Item 'PLC_CwuTemp' changed from 32.61357116699219 to 32.6171875
```

docbender, 2020-09-28:
To me it looks like address problem for some float number channel.  Don't you have there some channel for real number with byte or word address? Somethink like MB10F or MW10F. Only DWORD is supported. But I must fix validation procedure. It allows also BYTE and WORD now.

grzegorz914, 2020-09-28:
Yes,

U are right, some real Channels uses MBxxF address, removed it and start working correct. Need to be fixed.

docbender, 2020-09-29:
Address validation fixed

closed 2020-09-29 by docbender

grzegorz914, 2020-09-29:
after update to beta.2

`2020-09-29 12:33:48.034 [INFO ] [marthome.event.ItemStateChangedEvent] - Item 'TunerSat_Lastexecution' changed from 2020-09-29T12:33:32.922282+0200 to 2020-09-29T12:33:48.023120+0200
2020-09-29 12:33:56.829 [INFO ] [me.event.ThingStatusInfoChangedEvent] - Thing 'simatic:generic_device:635e89005f' changed from UNINITIALIZED (HANDLER_INITIALIZING_ERROR): null to UNINITIALIZED (HANDLER_MISSING_ERROR)
2020-09-29 12:33:56.843 [INFO ] [me.event.ThingStatusInfoChangedEvent] - Thing 'simatic:generic_device:635e89005f' changed from UNINITIALIZED (HANDLER_INITIALIZING_ERROR): null to INITIALIZING

==> /var/log/openhab/openhab.log <==
2020-09-29 12:33:56.852 [ERROR] [nal.common.AbstractInvocationHandler] - An error occurred while calling method 'ThingHandler.initialize()' on 'org.openhab.binding.simatic.internal.handler.SimaticGenericHandler@18a7a47': null
java.lang.NumberFormatException: null
        at java.lang.Integer.parseInt(Integer.java:614) ~[?:?]
        at java.lang.Integer.parseInt(Integer.java:770) ~[?:?]
        at org.openhab.binding.simatic.internal.simatic.SimaticChannel.checkAddress(SimaticChannel.java:113) ~[?:?]
        at org.openhab.binding.simatic.internal.simatic.SimaticChannel.init(SimaticChannel.java:85) ~[?:?]
        at org.openhab.binding.simatic.internal.handler.SimaticGenericHandler.initialize(SimaticGenericHandler.java:77) ~[?:?]
        at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:?]
        at jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[?:?]
        at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
        at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
        at org.openhab.core.internal.common.AbstractInvocationHandler.invokeDirect(AbstractInvocationHandler.java:152) [bundleFile:?]
        at org.openhab.core.internal.common.Invocation.call(Invocation.java:52) [bundleFile:?]
        at java.util.concurrent.FutureTask.run(FutureTask.java:264) [?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) [?:?]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) [?:?]
        at java.lang.Thread.run(Thread.java:834) [?:?]

==> /var/log/openhab/events.log <==
2020-09-29 12:33:56.857 [INFO ] [me.event.ThingStatusInfoChangedEvent] - Thing 'simatic:generic_device:635e89005f' changed from INITIALIZING to UNINITIALIZED (HANDLER_INITIALIZING_ERROR): null

==> /var/log/openhab/openhab.log <==
2020-09-29 12:33:56.857 [ERROR] [core.thing.internal.ThingManagerImpl] - Exception occurred while initializing handler of thing 'simatic:generic_device:635e89005f': null
java.lang.NumberFormatException: null
        at java.lang.Integer.parseInt(Integer.java:614) ~[?:?]
        at java.lang.Integer.parseInt(Integer.java:770) ~[?:?]
        at org.openhab.binding.simatic.internal.simatic.SimaticChannel.checkAddress(SimaticChannel.java:113) ~[?:?]
        at org.openhab.binding.simatic.internal.simatic.SimaticChannel.init(SimaticChannel.java:85) ~[?:?]
        at org.openhab.binding.simatic.internal.handler.SimaticGenericHandler.initialize(SimaticGenericHandler.java:77) ~[?:?]
        at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:?]
        at jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[?:?]
        at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
        at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
        at org.openhab.core.internal.common.AbstractInvocationHandler.invokeDirect(AbstractInvocationHandler.java:152) [bundleFile:?]
        at org.openhab.core.internal.common.Invocation.call(Invocation.java:52) [bundleFile:?]
        at java.util.concurrent.FutureTask.run(FutureTask.java:264) [?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) [?:?]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) [?:?]
        at java.lang.Thread.run(Thread.java:834) [?:?]

==> /var/log/openhab/events.log <==
2020-09-29 12:33:56.885 [INFO ] [smarthome.event.ThingUpdatedEvent   ] - Thing 'simatic:generic_device:635e89005f' has been updated.
2020-09-29 12:33:57.946 [INFO ] [marthome.event.ItemStateChangedEvent] - Item 'Meraki_SiecGoscie_Run' changed from OFF to ON
2020-09-29 12:33:58.736 [INFO ] [marthome.event.ItemStateChangedEvent] - Item 'Meraki_SiecGoscie_Run' changed from ON to OFF`