docbender/openHAB-Simatic #20: switch Channel set all 8 bytes to ON and report wrong state

grzegorz914, 2020-09-29:
H,

problem with Channel(switch), I have set stateAddress=DB6.DBX42.0 and commandAddres=DB6.DBX42.0 and toggle the correspondent switch then PLC set all 8bytes (0,1,2,3,4,5,6,7) to ON. Also the state is wron reported for the address.

grzegorz914, 2020-09-30:
I get also some warning during OH start, looks like **M** data type is not supported by **chNumber**:

`2020-09-30 18:20:26.143 [WARN ] [ternal.handler.SimaticGenericHandler] - Simatic PLC (iDom) - channel configuration error ChID=M8_0,StateAddress=M8.0,CmdAddress=null, Error=Unsupported address 'M8.0' for typeID=chNumber. Supported types B,W,D. Address example IB10, MW100, DB1.DBD0, DB1.DBD0F
2020-09-30 18:20:26.145 [WARN ] [ternal.handler.SimaticGenericHandler] - Simatic PLC (iDom) - channel configuration error ChID=M8_1,StateAddress=M8.1,CmdAddress=null, Error=Unsupported address 'M8.1' for typeID=chNumber. Supported types B,W,D. Address example IB10, MW100, DB1.DBD0, DB1.DBD0F
2020-09-30 18:20:26.147 [WARN ] [ternal.handler.SimaticGenericHandler] - Simatic PLC (iDom) - channel configuration error ChID=M8_2,StateAddress=M8.2,CmdAddress=null, Error=Unsupported address 'M8.2' for typeID=chNumber. Supported types B,W,D. Address example IB10, MW100, DB1.DBD0, DB1.DBD0F
2020-09-30 18:20:26.149 [WARN ] [ternal.handler.SimaticGenericHandler] - Simatic PLC (iDom) - channel configuration error ChID=M8_3,StateAddress=M8.3,CmdAddress=null, Error=Unsupported address 'M8.3' for typeID=chNumber. Supported types B,W,D. Address example IB10, MW100, DB1.DBD0, DB1.DBD0F
2020-09-30 18:20:26.150 [WARN ] [ternal.handler.SimaticGenericHandler] - Simatic PLC (iDom) - channel configuration error ChID=M8_4,StateAddress=M8.4,CmdAddress=null, Error=Unsupported address 'M8.4' for typeID=chNumber. Supported types B,W,D. Address example IB10, MW100, DB1.DBD0, DB1.DBD0F
2020-09-30 18:20:26.152 [WARN ] [ternal.handler.SimaticGenericHandler] - Simatic PLC (iDom) - channel configuration error ChID=M8_5,StateAddress=M8.5,CmdAddress=null, Error=Unsupported address 'M8.5' for typeID=chNumber. Supported types B,W,D. Address example IB10, MW100, DB1.DBD0, DB1.DBD0F
2020-09-30 18:20:26.158 [WARN ] [ternal.handler.SimaticGenericHandler] - Simatic PLC (iDom) - channel configuration error ChID=M8_7,StateAddress=M8.7,CmdAddress=null, Error=Unsupported address 'M8.7' for typeID=chNumber. Supported types B,W,D. Address example IB10, MW100, DB1.DBD0, DB1.DBD0F
2020-09-30 18:20:26.159 [WARN ] [ternal.handler.SimaticGenericHandler] - Simatic PLC (iDom) - channel configuration error ChID=M9_0,StateAddress=M9.0,CmdAddress=null, Error=Unsupported address 'M9.0' for typeID=chNumber. Supported types B,W,D. Address example IB10, MW100, DB1.DBD0, DB1.DBD0F
2020-09-30 18:20:26.161 [WARN ] [ternal.handler.SimaticGenericHandler] - Simatic PLC (iDom) - channel configuration error ChID=M9_1,StateAddress=M9.1,CmdAddress=null, Error=Unsupported address 'M9.1' for typeID=chNumber. Supported types B,W,D. Address example IB10, MW100, DB1.DBD0, DB1.DBD0F
2020-09-30 18:20:26.163 [WARN ] [ternal.handler.SimaticGenericHandler] - Simatic PLC (iDom) - channel configuration error ChID=M4_7,StateAddress=M4.7,CmdAddress=M4.7, Error=Unsupported address 'M4.7' for typeID=chNumber. Supported types B,W,D. Address example IB10, MW100, DB1.DBD0, DB1.DBD0F`

docbender, 2020-10-01:
Fixed datablock bit address for Switch or Contact in  71e1ebb7c7b57d0cbe2605942ee0160f7fa00ed0

For second question. Number is now only supported for BYTE, WORD, DWORD. Is it useful to have it also for BIT?

grzegorz914, 2020-10-01:
Right now I have changed chNumber to chContact but this operation inverted the reported state from PLC. I use some M1.x area report alarms 0/1 inactive/active. It’s not important for me I can change this in PLC code. Was only as question because in old binding this was supported.

Many thanks for Your god support.

closed 2020-10-01 by grzegorz914

docbender, 2020-10-01:
I can add bit support. I did not know that anybody could use it

docbender, 2020-10-01:
You are right. Bit was supported in OH1 binding. So I implemented it in OH3 too. Check new release

grzegorz914, 2020-10-01:
👍