docbender/openHAB-Simatic #23: Somtimes get error

grzegorz914, 2020-11-08:
Hi,
I get sometimes this error after send value to PLC:

`2020-11-08 11:11:34.353 [ERROR] [nternal.simatic.SimaticGenericDevice] - 192.168.1.30 - 
	at org.openhab.binding.simatic.internal.simatic.SimaticWriteDataArea.create(SimaticWriteDataArea.java:48) ~[bundleFile:?]
	at org.openhab.binding.simatic.internal.simatic.SimaticGenericDevice.sendData(SimaticGenericDevice.java:287) [bundleFile:?]
	at org.openhab.binding.simatic.internal.handler.SimaticGenericHandler.handleCommand(SimaticGenericHandler.java:178) [bundleFile:?]
	at jdk.internal.reflect.GeneratedMethodAccessor144.invoke(Unknown Source) ~[?:?]
	at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
	at org.openhab.core.internal.common.AbstractInvocationHandler.invokeDirect(AbstractInvocationHandler.java:154) [bundleFile:?]
	at org.openhab.core.internal.common.InvocationHandlerSync.invoke(InvocationHandlerSync.java:59) [bundleFile:?]
	at org.openhab.core.thing.internal.profiles.ProfileCallbackImpl.handleCommand(ProfileCallbackImpl.java:80) [bundleFile:?]
	at org.openhab.core.thing.internal.profiles.SystemDefaultProfile.onCommandFromItem(SystemDefaultProfile.java:48) [bundleFile:?]
	at jdk.internal.reflect.GeneratedMethodAccessor142.invoke(Unknown Source) ~[?:?]
	at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
	at org.openhab.core.internal.common.AbstractInvocationHandler.invokeDirect(AbstractInvocationHandler.java:154) [bundleFile:?]
	at org.openhab.core.internal.common.Invocation.call(Invocation.java:52) [bundleFile:?]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) [?:?]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) [?:?] `

docbender, 2020-11-09:
Look like commanded channel does not have specified a command address in configuration. It is not clear in log file. I'll fix it to make it clearer.

docbender, 2020-11-09:
Better command address check is now present in 3.0.1 release.

closed 2020-11-09 by docbender

grzegorz914, 2020-11-09:
Thanks it working with correct Warning now, but it is possible that U add in channel config enhancement  switch with state ON/OFF Command. In OFF state the warning will not be displayed.?

docbender, 2020-11-09:
Why "Command ON/OFF" switch.? If you did not specify address it is as you turn OFF the switch.

grzegorz914, 2020-11-09:
Yes U are right but get always WARNING in log.

docbender, 2020-11-09:
I can suppress this WARNING to INFO or maybe better print this message only once in item livetime. On other hand best way is not to set items command without command address :-)

grzegorz914, 2020-11-09:
Ok,

better move to INFO