docbender/openHAB-Simatic #24: OH3

grzegorz914, 2020-11-21:
Hi,

looks like send Number value to the PLC not working:

After set some value for rollerschutter position and running the rule nothing happens on the PLC site:
Here the rule:

```
rule "Roleta biuro"
when
    Item PietroBiuroRoleta received command
then
    if (!(receivedCommand instanceof Number))
        switch receivedCommand {
            case UP :    if (PLC_PietroBiuroRoletaStany.state <=1) {
                                 PietroBiuroRoleta.sendCommand(STOP) }
                            else
                                { PLC_PietroBiuroRoletaOtworz.sendCommand(ON)
                                 PLC_PietroBiuroRoletaOtworz.sendCommand(OFF) }
            case DOWN :  if (PLC_PietroBiuroRoletaStany.state <=1) {
                                 PietroBiuroRoleta.sendCommand(STOP) }  
                            else 
                                { PLC_PietroBiuroRoletaZamknij.sendCommand(ON)
                                 PLC_PietroBiuroRoletaZamknij.sendCommand(OFF) }
            case STOP :  if (PLC_PietroBiuroRoletaStany.state == 0) { 
                             PLC_PietroBiuroRoletaOtworz.sendCommand(ON)
                             PLC_PietroBiuroRoletaOtworz.sendCommand(OFF) }  
                         else if (PLC_PietroBiuroRoletaStany.state == 1) { 
                             PLC_PietroBiuroRoletaZamknij.sendCommand(ON)
                             PLC_PietroBiuroRoletaZamknij.sendCommand(OFF) }     
        }
    else
        if (receivedCommand instanceof Number && (PLC_PietroBiuroRoletaStany.state >= 2))
                       PLC_PietroBiuroRoletaPozycjaZ.sendCommand(receivedCommand)
end
```

Log:
```
2020-11-21 12:35:53.651 [INFO ] [smarthome.event.ItemCommandEvent    ] - Item 'PietroBiuroRoleta' received command 57
2020-11-21 12:35:54.702 [INFO ] [smarthome.event.ItemCommandEvent    ] - Item 'PLC_PietroBiuroRoletaPozycjaZ' received command 57
2020-11-21 12:35:54.717 [INFO ] [marthome.event.ItemStateChangedEvent] - Item 'PietroBiuroRoleta' changed from 81 to 57
2020-11-21 12:35:54.877 [INFO ] [rthome.event.ItemStatePredictedEvent] - Item 'PLC_PietroBiuroRoletaPozycjaZ' predicted to become 57
2020-11-21 12:35:54.927 [INFO ] [marthome.event.ItemStateChangedEvent] - Item 'PLC_PietroBiuroRoletaPozycjaZ' changed from 0 to 57
2020-11-21 12:35:55.216 [INFO ] [marthome.event.ItemStateChangedEvent] - Item 'PLC_PietroBiuroRoletaPozycjaZ' changed from 57 to 0
```

also same behaviour for set Thermostat temperature but here the item is connected direct with PLC channel without rule:
```
2020-11-21 12:43:08.149 [INFO ] [smarthome.event.ItemCommandEvent    ] - Item 'BiuroTermTrybZadanySlowo' received command Chłodzenie
2020-11-21 12:43:10.167 [INFO ] [smarthome.event.ItemCommandEvent    ] - Item 'PLC_BiuroTempZ' received command 22.0
2020-11-21 12:43:10.311 [INFO ] [rthome.event.ItemStatePredictedEvent] - Item 'PLC_BiuroTempZ' predicted to become 22.0
2020-11-21 12:43:10.373 [INFO ] [marthome.event.ItemStateChangedEvent] - Item 'PLC_BiuroTempZ' changed from 21.0 °C to 22.0 °C
2020-11-21 12:43:11.248 [INFO ] [marthome.event.ItemStateChangedEvent] - Item 'PLC_BiuroTempZ' changed from 22.0 °C to 21.0 °C
```

docbender, 2020-11-21:
Hi,
To be clear. It works normally for some time, then it stops for unknown reason?

grzegorz914, 2020-11-21:
No, it never worked

docbender, 2020-11-21:
Strange. My test OH3 running 10days in row without any write problem. Right now just added rollershutter channel and it works as expected.
Nothing in openhab.log? To find out whats going on I need your debug log.

grzegorz914, 2020-11-21:
Please try to use Number Channel not Rollershutter Channel.

grzegorz914, 2020-11-21:
Here the TRACE:

`2020-11-21 22:13:12.790 [DEBUG] [ternal.handler.SimaticGenericHandler] - Simatic PLC (iDom) - Command 0.71000000(class org.openhab.core.library.types.DecimalType) for channel simatic:generic_device:635e89005f:DB7_DBB105
2020-11-21 22:13:12.791 [DEBUG] [nternal.simatic.SimaticGenericDevice] - 192.168.1.30 - Adding command into queue
2020-11-21 22:13:12.792 [DEBUG] [nternal.simatic.SimaticGenericDevice] - 192.168.1.30 - Processing commandQueue - length 1. Thread=3423
2020-11-21 22:13:12.793 [DEBUG] [.simatic.internal.simatic.SimaticTCP] - 192.168.1.30 - writeBytes(area=132,db=7,adr=105,len=1,datalen=1,data=0x0,)
2020-11-21 22:13:12.799 [TRACE] [nternal.simatic.SimaticGenericDevice] - 192.168.1.30 - Incoming data - item:simatic:generic_device:635e89005f:MD72/state:66.5999984741211 `

grzegorz914, 2020-11-22:
Set Target Thermostat Temperature fixed with removed Type:Temperature from Item.
Set Target Position for Blind not solved, it’s not matter what Number Value is Set, always send 0 zero to the PLC.

grzegorz914, 2020-11-22:
Hi,

some DEBUG:

```
2020-11-22 14:12:45.388 [INFO ] [smarthome.event.ItemCommandEvent    ] - Item 'PLC_PietroBiuroRoletaPozycjaZ' received command 67
2020-11-22 14:12:45.390 [DEBUG] [ternal.handler.SimaticGenericHandler] - Simatic PLC (iDom) - Command 0.67000000(class org.openhab.core.library.types.DecimalType) for channel simatic:generic_device:635e89005f:DB7_DBB105
2020-11-22 14:12:45.393 [DEBUG] [nternal.simatic.SimaticGenericDevice] - 192.168.1.30 - Adding command into queue
2020-11-22 14:12:45.394 [DEBUG] [nternal.simatic.SimaticGenericDevice] - 192.168.1.30 - Processing commandQueue - length 1. Thread=1339
2020-11-22 14:12:45.395 [DEBUG] [.simatic.internal.simatic.SimaticTCP] - 192.168.1.30 - writeBytes(area=132,db=7,adr=105,len=1,datalen=1,data=0x0,)
```
as U see always try to send 0

grzegorz914, 2020-11-22:
I have fixed this adding to the part of rule **receivedCommand as Number**

```
if (receivedCommand instanceof Number && (PLC_PietroBiuroRoletaStany.state >= 2))
                       PLC_PietroBiuroRoletaPozycjaZ.sendCommand(receivedCommand as Number)
```

docbender, 2020-11-22:
Looks like OH internally converted PercentType to DecimalType in range 0.0-1.0. So it was everytime 0.

grzegorz914, 2020-11-22:
OK, 
may be bug in OH3 in OH2.x was OK.

I close this as solved.

grzegorz914, 2021-01-08:
I think is not resolved again, using Item with Type get error: #25

docbender, 2021-01-11:
Number:Temperature issue item fixed in #25

closed 2021-01-11 by docbender