docbender/openHAB-Simatic #25: Number value does not send if use with type...

grzegorz914, 2021-01-08:
Hi,

i have observed if we use item with type like **Number:Temperature** then binding always get error with, this is also related to #24:

```
==> /var/log/openhab/openhab.log <==
2021-01-08 10:16:07.803 [ERROR] [nternal.simatic.SimaticGenericDevice] - 192.168.1.30 - ChannelUID=simatic:generic_device:635e89005f:DB6_DBD52. Cannot create WriteDataArea. Command for ChannelType=chNumber must be DecimalType.
java.lang.Exception: Cannot create WriteDataArea. Command for ChannelType=chNumber must be DecimalType
        at org.openhab.binding.simatic.internal.simatic.SimaticWriteDataArea.create(SimaticWriteDataArea.java:55) ~[bundleFile:?]
        at org.openhab.binding.simatic.internal.simatic.SimaticGenericDevice.sendData(SimaticGenericDevice.java:285) [bundleFile:?]
        at org.openhab.binding.simatic.internal.handler.SimaticGenericHandler.handleCommand(SimaticGenericHandler.java:187) [bundleFile:?]
        at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:?]
        at jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[?:?]
        at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
        at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
        at org.openhab.core.internal.common.AbstractInvocationHandler.invokeDirect(AbstractInvocationHandler.java:154) [bundleFile:?]
        at org.openhab.core.internal.common.InvocationHandlerSync.invoke(InvocationHandlerSync.java:59) [bundleFile:?]
        at com.sun.proxy.$Proxy429.handleCommand(Unknown Source) [?:?]
        at org.openhab.core.thing.internal.profiles.ProfileCallbackImpl.handleCommand(ProfileCallbackImpl.java:80) [bundleFile:?]
        at org.openhab.core.thing.internal.profiles.SystemDefaultProfile.onCommandFromItem(SystemDefaultProfile.java:48) [bundleFile:?]
        at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:?]
        at jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[?:?]
        at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
        at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
        at org.openhab.core.internal.common.AbstractInvocationHandler.invokeDirect(AbstractInvocationHandler.java:154) [bundleFile:?]
        at org.openhab.core.internal.common.Invocation.call(Invocation.java:52) [bundleFile:?]
        at java.util.concurrent.FutureTask.run(FutureTask.java:264) [?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) [?:?]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) [?:?]
        at java.lang.Thread.run(Thread.java:834) [?:?]
```

docbender, 2021-01-10:
Number item is restricted for openHAB DecimalType only. Obviously OH try to write something different. There are some questions?
How can I reproduce this issue? How is the item written? Through GUI or script? How exactly looks item definition?

grzegorz914, 2021-01-10:
On first screen U see created item through  GUI as **Number:Temperature**:
[image]
Here is the Channel config:
[image]
Herę is Item linked with Channel:
[image]

if item is as **Number** all working as excepted but if set additional **Item Quanity Type Temperature** then always get reported error.

docbender, 2021-01-11:
Problem is fixed. I did not count with QuantityType.  You can try new release.

closed 2021-01-11 by docbender

grzegorz914, 2021-01-11:
Thanks,

all working correct now.