docbender/openHAB-Simatic #26: connection lost

mscifm, 2021-01-14:
Hi all,

Had installed a fresh debian 10 with openhab3 and receive the following error. I followed the installation instructions on website

2021-01-14 21:40:59.664 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'simatic:generic_device:8b6542dbda:f228471487' changed from OFFLINE (BRIDGE_OFFLINE) to ONLINE

==> /var/log/openhab/openhab.log <==
2021-01-14 21:44:50.083 [ERROR] [nternal.simatic.SimaticGenericDevice] - 192.168.25.101 -
org.openhab.binding.simatic.internal.simatic.SimaticReadException: Read data area error (Area=M0-M1, Return code=0xffffff7d, Error=No data returned. Connection lost?})
        at org.openhab.binding.simatic.internal.simatic.SimaticTCP.readDataArea(SimaticTCP.java:309) ~[bundleFile:?]
        at org.openhab.binding.simatic.internal.simatic.SimaticGenericDevice.checkNewData(SimaticGenericDevice.java:425) [bundleFile:?]
        at org.openhab.binding.simatic.internal.simatic.SimaticGenericDevice.execute(SimaticGenericDevice.java:159) [bundleFile:?]
        at org.openhab.binding.simatic.internal.simatic.SimaticGenericDevice.lambda$0(SimaticGenericDevice.java:113) [bundleFile:?]
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515) [?:?]
        at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:305) [?:?]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:305) [?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) [?:?]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) [?:?]
        at java.lang.Thread.run(Thread.java:834) [?:?]
2021-01-14 21:44:51.094 [INFO ] [nternal.simatic.SimaticGenericDevice] - 192.168.25.101 - Trying to reconnect

==> /var/log/openhab/events.log <==
2021-01-14 21:44:51.096 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'simatic:bridge:8b6542dbda' changed from ONLINE to OFFLINE (COMMUNICATION_ERROR)
2021-01-14 21:44:51.100 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'simatic:generic_device:8b6542dbda:f228471487' changed from ONLINE to OFFLINE (BRIDGE_OFFLINE)
2021-01-14 21:44:51.101 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'simatic:generic_device:2f77a90a34' changed from ONLINE to OFFLINE (BRIDGE_OFFLINE)

docbender, 2021-01-15:
There is no response from PLC. I'm expecting that configuration (rack,slot) is correct. Is not there any network issue? What is ping response? Is any data from PLC read correctly?

mscifm, 2021-01-15:
Hi, continious PING is OK, in the case of error I´ve got ping losts. There is no interaction with the plc or the openhab instance at the error-moment

docbender, 2021-01-15:
If there is ping lost, it looks like network issue. Maybe problem with some ethernet connector

mscifm, 2021-01-15:
As I wrote before, the PING lost appears in the case of the openhab error.

th3elk, 2021-01-20:
What type of plc do you use?
For S7-1500 or S7-1200 you have to allow Put/Get-communication in the hardware-settings of the plc.

docbender, 2021-02-03:
Closing due to lack of feedback

closed 2021-02-03 by docbender