docbender/openHAB-Simatic #27: recurring connections issue

mscifm, 2021-01-14:
Hi all,

Had installed a fresh debian 10 with openhab3 and receive the following error. I followed the installation instructions on website

2021-01-14 21:40:59.664 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'simatic:generic_device:8b6542dbda:f228471487' changed from OFFLINE (BRIDGE_OFFLINE) to ONLINE

==> /var/log/openhab/openhab.log <==
2021-01-14 21:44:50.083 [ERROR] [nternal.simatic.SimaticGenericDevice] - 192.168.25.101 -
org.openhab.binding.simatic.internal.simatic.SimaticReadException: Read data area error (Area=M0-M1, Return code=0xffffff7d, Error=No data returned. Connection lost?})
        at org.openhab.binding.simatic.internal.simatic.SimaticTCP.readDataArea(SimaticTCP.java:309) ~[bundleFile:?]
        at org.openhab.binding.simatic.internal.simatic.SimaticGenericDevice.checkNewData(SimaticGenericDevice.java:425) [bundleFile:?]
        at org.openhab.binding.simatic.internal.simatic.SimaticGenericDevice.execute(SimaticGenericDevice.java:159) [bundleFile:?]
        at org.openhab.binding.simatic.internal.simatic.SimaticGenericDevice.lambda$0(SimaticGenericDevice.java:113) [bundleFile:?]
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515) [?:?]
        at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:305) [?:?]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:305) [?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) [?:?]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) [?:?]
        at java.lang.Thread.run(Thread.java:834) [?:?]
2021-01-14 21:44:51.094 [INFO ] [nternal.simatic.SimaticGenericDevice] - 192.168.25.101 - Trying to reconnect

==> /var/log/openhab/events.log <==
2021-01-14 21:44:51.096 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'simatic:bridge:8b6542dbda' changed from ONLINE to OFFLINE (COMMUNICATION_ERROR)
2021-01-14 21:44:51.100 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'simatic:generic_device:8b6542dbda:f228471487' changed from ONLINE to OFFLINE (BRIDGE_OFFLINE)
2021-01-14 21:44:51.101 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'simatic:generic_device:2f77a90a34' changed from ONLINE to OFFLINE (BRIDGE_OFFLINE)

mscifm, 2021-01-14:
OK, additionally , error jumps from M0-M1 to Q0-Q1

==> /var/log/openhab/openhab.log <==
2021-01-14 21:52:30.099 [ERROR] [nternal.simatic.SimaticGenericDevice] - 192.168.25.101 -
org.openhab.binding.simatic.internal.simatic.SimaticReadException: Read data area error (Area=Q0-Q1, Return code=0xffffff7d, Error=No data returned. Connection lost?})
        at org.openhab.binding.simatic.internal.simatic.SimaticTCP.readDataArea(SimaticTCP.java:309) ~[bundleFile:?]
        at org.openhab.binding.simatic.internal.simatic.SimaticGenericDevice.checkNewData(SimaticGenericDevice.java:425) [bundleFile:?]
        at org.openhab.binding.simatic.internal.simatic.SimaticGenericDevice.execute(SimaticGenericDevice.java:159) [bundleFile:?]
        at org.openhab.binding.simatic.internal.simatic.SimaticGenericDevice.lambda$0(SimaticGenericDevice.java:113) [bundleFile:?]
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515) [?:?]
        at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:305) [?:?]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:305) [?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) [?:?]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) [?:?]
        at java.lang.Thread.run(Thread.java:834) [?:?]
2021-01-14 21:52:31.101 [INFO ] [nternal.simatic.SimaticGenericDevice] - 192.168.25.101 - Trying to reconnect

==> /var/log/openhab/events.log <==
2021-01-14 21:52:31.103 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'simatic:bridge:8b6542dbda' changed from ONLINE to OFFLINE (COMMUNICATION_ERROR)
2021-01-14 21:52:31.108 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'simatic:generic_device:2f77a90a34' changed from ONLINE to OFFLINE (BRIDGE_OFFLINE)
2021-01-14 21:52:31.109 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'simatic:generic_device:8b6542dbda:f228471487' changed from ONLINE to OFFLINE (BRIDGE_OFFLINE)

==> /var/log/openhab/openhab.log <==
2021-01-14 21:53:02.594 [ERROR] [.simatic.internal.simatic.SimaticTCP] - 192.168.25.101 - cannot connect to PLC. Die Wartezeit für die Verbindung ist abgelaufen (Connection timed out)
2021-01-14 21:53:02.595 [INFO ] [nternal.simatic.SimaticGenericDevice] - 192.168.25.101 - Trying to reconnect
2021-01-14 21:53:03.631 [INFO ] [.simatic.internal.simatic.SimaticTCP] - 192.168.25.101 - connected. PDU size = 240B

==> /var/log/openhab/events.log <==
2021-01-14 21:53:03.634 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'simatic:bridge:8b6542dbda' changed from OFFLINE (COMMUNICATION_ERROR) to ONLINE
2021-01-14 21:53:03.636 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'simatic:generic_device:2f77a90a34' changed from OFFLINE (BRIDGE_OFFLINE) to ONLINE
2021-01-14 21:53:03.638 [INFO ] [ab.event.ThingStatusInfoChangedEvent] - Thing 'simatic:generic_device:8b6542dbda:f228471487' changed from OFFLINE (BRIDGE_OFFLINE) to ONLINE

docbender, 2021-01-15:
Related #26

closed 2021-01-15 by docbender