docbender/openHAB-Simatic #29: Problem with Rollershutter

Maddin2020, 2021-04-05:
Hi Docbender, 

i use your great binding in Openhab 3. Now i have a little problem with the Rollershutter. I have created a Rollershutter item in my things file. 

`    Thing generic_device Gaestebad "S7 SPS - 101 - EG Gästebad" {
        Channels:                 
            Type chRollershutter: Jalousie              "Jalousie"                      [ stateAddress="DB2001.DBB104", commandAddress="DB2001.DBB104" ]                
    } `

At in the items file 

`Rollershutter Jalousie "Jalousie" {channel="simatic:generic_device:S7-Home:Gaestebad:Jalousie"}`

But if i save the items file i will see the follow error in the log and the item status is "NULL"

2021-04-05 20:33:49.394 [WARN ] [ternal.handler.SimaticGenericHandler] - S7 SPS - 101 - EG Gästebad - cannot get value to refresh. Channel simatic:generic_device:S7-Home:Gaestebad:Jalousie not found.

docbender, 2021-04-06:
Hi,

try to delete your description "Jalousie" from channel configuration. Should looks like:

    Channels: Type chRollershutter: Jalousie  [ ... ]

And remember, that command and status address should be different

MFI-Net, 2021-04-07:
Hi, has somebody a sample how Items should look like withthe new Simatic binding. 

Thanks in advance

docbender, 2021-04-07:
Like any other items in OH3. See https://www.openhab.org/docs/configuration/items.html

    Number   item1 "description" { channel="simatic:generic_device:...." }

Concreate channel ID you can find in the channel configuration

Maddin2020, 2021-04-07:
Thanks Docbender i think i must change my plc software.

closed 2021-05-13 by docbender