docbender/openHAB-Simatic #30: Binding stop working with latest OH snapshots

grzegorz914, 2021-05-12:
Hi,

in latest OH snapshots The binding stop working with:

```
org.osgi.framework.BundleException: Could not resolve module: org.openhab.binding.simatic [265]
  Unresolved requirement: Import-Package: javax.measure; version="[1.0.0,2.0.0)"
        at org.eclipse.osgi.container.Module.start(Module.java:463) ~[org.eclipse.osgi-3.16.200.jar:?]
        at org.eclipse.osgi.internal.framework.EquinoxBundle.start(EquinoxBundle.java:440) ~[org.eclipse.osgi-3.16.200.jar:?]
        at org.apache.felix.fileinstall.internal.DirectoryWatcher.startBundle(DirectoryWatcher.java:1260) [bundleFile:3.6.8]
        at org.apache.felix.fileinstall.internal.DirectoryWatcher.startBundles(DirectoryWatcher.java:1233) [bundleFile:3.6.8]
        at org.apache.felix.fileinstall.internal.DirectoryWatcher.doProcess(DirectoryWatcher.java:520) [bundleFile:3.6.8]
        at org.apache.felix.fileinstall.internal.DirectoryWatcher.process(DirectoryWatcher.java:365) [bundleFile:3.6.8]
        at org.apache.felix.fileinstall.internal.DirectoryWatcher.run(DirectoryWatcher.java:316) [bundleFile:3.6.8]

==> /var/log/openhab/events.log <==
```

docbender, 2021-05-13:
Package javax.measure was updated to higher version. I will prepare new OH3.1 binding release

grzegorz914, 2021-05-13:
OK,

Thanks

docbender, 2021-05-14:
I released 3.1 version. It is able to run with latest OH 3.1 snapshots (tested with build #2380). But there are still problems with new javax.measure API (lot of javax.measure.format.MeasurementParseException in log). See https://github.com/openhab/openhab-core/pull/2319 for details

Use with care.

grzegorz914, 2021-05-22:
Hi,

it working fine now.

Thanks

closed 2021-05-22 by grzegorz914