docbender/openHAB-Simatic #33: Performance Issues - CPU/Memory

Willem-Renders, 2021-09-24:
Hi,

First of all, much love for this addon. Helped me a lot with my home automation system.

I run my openhab in a docker container on a RPI4.
And sometimes (1-3 times a month) my CPU starts to spike and I can see a high memory use.

I have logged this on the forums if there is a way to find the issue:
https://community.openhab.org/t/oh3-ramping-cpu-issue/126692 

By analyzing I have found out that the simatic addon is taking a lot of CPU and keeps ramping up, same as memory.
I have around +-185 items connected to a simatic state. Polling is default and all of them using the DB addresses except 7 items that read state of an I address.

[image]

Before red is with simatic, between red and orange I was trying to set a different polling rate.
This didn't work, if I set a polling of +2s my switches would go on, but I couldn't turn them off.
After orange is without simatic addon.

System Info:
- Hardware: Raspberry Pi 4B - 4 Gb - 2.4 GHz
- Runtime: Docker Engine - Community Version 19.03.13
- Openhab: 3.2.0.M2 (Also happened on 3.1.0.M5)

docbender, 2021-11-01:
Hi,

I don't know what standard OH behavior in Docker looks like. But from your CPU usage it looks like CPU load is ramping little bit even without Simatic binding.

However, I've put binding to the test with 3000 PLC items and I did not find any memory leak. My test configuration:
* OH 3.2.0 M3 standalone
* OpenJDK 11.0.11

What I found out is that value update is for OH very resource intensive (lot of memory allocation and associated CPU load). Even if item value did not changed. 
So I added into binding value comparison. With enabled comparison only changed value is sent into OH. If the values are not changed frequently, the load is significantly lower.
[image]
This behavior can be changed in configuration as _Value update mode_. Two modes are supported:
* _OnChange_ (only changes) 
* _Poll_ (everything). 

You can try binding with this new features [https://github.com/docbender/openHAB-Simatic/releases/download/3.2.0-beta.1/org.openhab.binding.simatic-3.2.0-SNAPSHOT.jar](https://github.com/docbender/openHAB-Simatic/releases/download/3.2.0-beta.1/org.openhab.binding.simatic-3.2.0-SNAPSHOT.jar)

Willem-Renders, 2021-11-01:
Hi Docbender,

Thanks a lot for the in depth analysis, I will install the new version later and try the new changes.
Will give you an update about that.

About the CPU and memory, it is true that simantic is taking quite alot of resources, hopefully the update will give improvements. However the CPU and memory ramping was something else and quite difficult to find.

So simantic is just taking alot of resource but is very stable, and then there was another addon which ramps the cpu after some time.

th3elk, 2021-11-01:
Hi docbender,

thanks for your great work! I would like to test the new version of the binding too, because sometimes my system based on RPI 4 also runs out of memory. I didn't find a solution yet, but maybe the new version of the binding could help.
Can you tell me where I have to put the configuration? Is it in the *.things file in the first line like this:

`Bridge simatic:bridge:S7-1500 "My_PLC"  [address="192.168.178.54", rack="0", slot="1", pollRate="50", charset="ISO-8859-1", **update mode="OnChange"** ] {...`
Thanks for your help!

th3elk, 2021-11-01:
I think I found it, it has to be:

`Bridge simatic:bridge:S7-1500 "My_PLC"  [address="192.168.178.54", rack="0", slot="1", pollRate="50", charset="ISO-8859-1", updatemode="OnChange"] {...`
If I'm right, ther should be no spacemark in updatemode.

At a firft look, it looks like my Items get updated more quickly, whem I trigger them. But I also did a reboot to use the new version of the binding. I will report in some days, if the chage hat an effect.

docbender, 2021-11-01:
Parameter name is "updateMode". If parameter is not specified, OnChange is set as default. I have to add it into documentation.
Btw. after new version of binding is installed, I recommend run "openhab-cli clean-cache"

th3elk, 2021-11-01:
Thanks for clarification.
I have one more question. I wanted to look at the logs of the simatic-binding as I did unter OH2. I used: log:set DEBUG org.openhab.binding.simatic
The only entrys I get are, when I change a value, I see how the data is send to PLC. But I don't have any entrys about reading parameters from PLC. I don't know when it changed, but is there a way to get the log-entrys of the readingparameters? Sometimes openhab doesn't update the states of my items. I want to check if it's a problem of openhab or the binding.

Willem-Renders, 2021-11-01:
Thanks Docbender, new versions does magic.

Average CPU was 25% before, after the update it is 4%

docbender, 2021-11-02:
> I wanted to look at the logs of the simatic-binding as I did unter OH2. I used: log:set DEBUG org.openhab.binding.simatic The only entrys I get are, when I change a value, I see how the data is send to PLC. But I don't have any entrys about reading parameters from PLC. 

As part of the performance tests and bottleneck search I commented out debug message "PLC - Reading finished. Area=DBxxx-DBxxx".  I will put it back in next release.

docbender, 2021-11-02:
> Average CPU was 25% before, after the update it is 4%

One of the most resource consumer is logging state events. If you haven't already, disable it (set _ItemStateEvent_ to _WARN_ level) is good idea too.

closed 2021-12-13 by github-actions[bot]