docbender/openHAB-Simatic #34: Binding stops working in different connection cases

NeurosProjekte, 2021-11-21:
Hello Docbender,
i found some issues with the binding when there are connection problems or after the restart of the complete openhab.

My setup:
- OpenHab in a docker container in the version 3.1.0 release build
- Simatic S7-300 PLC with CP for ethernet connection
- Everything with static IP-Adresses and wired with ethernet cables (no wireless connection)
- Main PLC as bridge in openhab
- Five generic devices, each for a DB in the PLC with different functions

Issue description:
In some cases the binding stops working. In the openhab bindings page it still shows "online" in the bridge and also on the devices, but it doesn't send or read data to and from the plc. I also got that issue sometimes on only one device and other ones still work as normal.

Cases where this does happen:
- On friday i made an firmware update on my router, so my network dns was down for a few minutes. After the server was online again, the binding doesn't get back to work but shows online.
- It also happend sometimes if i restart my openhab docker

So for me it looks like, if there was a connection issue during a data transfer, the binding then freezes and had to restart again manualy to get back working.

Fast solution for get it working again:
Pause and start the bridge first, then do the same thing on all the devices.

Maybe you could take a look into your binding and get this problem fixed. If i could help you with logs or something else, let me know what you need :-)

Thanks you for keeping this binding alive!

Best regards,
Marcel

docbender, 2021-11-24:
There was an issue where a scheduled task to read data from PLC was interrupted while trying to reconnect. PLC was sebsequently connected, but no data was read.

Issue is fixed now. You can try new release: 
https://github.com/docbender/openHAB-Simatic/releases/download/3.2.0-beta.2/org.openhab.binding.simatic-3.2.0-SNAPSHOT.jar

NeurosProjekte, 2021-12-11:
Looks like it worked perfectly now :-) Thanks for fixing this bug. Works since 15 days in my setup without any problems. Good job with this binding, not sure why it is not in the official OH bindings? :-)

closed 2021-12-23 by docbender