docbender/openHAB-Simatic #36: I can´t read the Data Type Time from PLC

Maddin2020, 2021-12-19:
Hello Docbender,

i want read a value with the Datatype "Time" from the PLC (32 bit). Example i want read the actual Time Value from a IEC Timer. Then i have connect the Data Point DB2001.DBD3028  in the Things Data Table. I only received 0 in the value Dammerung_Restzeitms.

All other data Points working correctly. 

```
 Thing generic_device gobal {
        Channels:                          
            Type chNumber: Dammerung_Restzeitms         "Dämmerung Restzeit"            [ stateAddress="DB2001.DBW3028"]
    }
``` 

Thanks a lot for your support and thank for the great binding. 

Martin

docbender, 2021-12-19:
Hi,
isn't problem in address "DB2001.DBW3028"? It should be DBDxx instead DBWxx. Right?

Maddin2020, 2021-12-19:
Hey docbender yes DBxxx.DBD is right because the data Type "Time" have 32 Bit. If i change the data Type in "DInt" then it work but i don't understand the reason because the  length is the same. I think you know that the data type Time is only the value in Milli seconds. If i select Time data Type in the Tia Portal i can read it easilier.

docbender, 2021-12-20:
Data are read as byte array. I have no idea why it is read as zero. If you read DBD address with whatever type on Simatic side, on binding side have to be data stored in that 4 bytes (number of millisecond in this case).
As a workaround I would use a transfer in code:
```
 L    TIME_VARIABLE
 T    DINT_VARIABLE
```

Maddin2020, 2021-12-20:
Thanks again for answer. Ok i will use this workaround. Do you have the same Issue if you Test it on your plc ?

docbender, 2021-12-21:
With S7-400 no problem
[image]

closed 2021-12-23 by docbender