docbender/openHAB-Simatic #39: Number value NULL after OH update/restart

grzegorz914, 2022-02-21:
Hi,

I have observed that every update/restart OH the Number value are not updated/aktualisiert on first run, still NULL as long as the value not changed on S7 or I’m restart the thing manually from UI.

docbender, 2022-02-23:
Looks like this is related to #38 issue. Values updated during startup are not transferred to the OH core. Maybe it is system configuration related. Is your OH PC slow or high performance?

grzegorz914, 2022-02-23:
Its not related to #38, the Simatic Versinon is never updated. All other internal channel like PDU, Areas, Request, Bytes are updated correct.
The system is Bulseye on RPI4 4GB.
May be the Simatic Bindig is loading before OH is ready to update this value.

grzegorz914, 2022-02-23:
Sorry, you are right is related to #38, after restart Simatic:bridge the value is updated.

grzegorz914, 2022-02-23:
I think the all values need to be forced update on first run in OnChange mode. 
The Version is never updated because is a static string and not changed after binding start. The other values after some time changed on S7 and are also updated in OH.

grzegorz914, 2022-03-09:
@docbender 
thanks for try to fix this but this PR with delay start does not fix the issue, If the thing is OnChange mode the value are not updated on first start and remain NULL as long as the values not changed on PLC side. 
I think you need force push all actual PLC values to the core after Simatic binding initialization complete (only on first cycle), next watching for value changed.

Thanks

docbender, 2022-03-09:
What about bridge item values? For example binding version? It is null too?

What about your hardware? Is your hardware slow?

I would need a log dump in OnScan mode to see when the data starts to transfer after PLC is connected. I'm interested in "openhab.event.ItemStateChangedEvent", where you can see value change from NULL.

grzegorz914, 2022-03-09:
Bridge same behaviour the version still NULL as long as I pause/start bridge.
My hardware is Raspberry Pi 4B+ 4GB, I think is not slow, CPU usage is 19%

docbender, 2022-03-09:
Please collect log after OH restart with org.openhab.binding.simatic set to DEBUG level and openhab.event.ItemStateChangedEvent set to INFO level.

grzegorz914, 2022-03-09:
Here the logs, interesant is that the channels with NULL value does not exist in the logs.
[events.log.zip](https://github.com/docbender/openHAB-Simatic/files/8214061/events.log.zip)
[openhab.log.zip](https://github.com/docbender/openHAB-Simatic/files/8214063/openhab.log.zip)

docbender, 2022-03-09:
Important records from your logs:

    2022-03-09 12:01:06.388 [INFO ] [org.openhab.ui.internal.UIService   ] - Started UI on port 8080
    2022-03-09 12:01:33.323 [INFO ] [ternal.handler.SimaticHandlerFactory] - Simatic binding (v.3.2.1) has been started.
    2022-03-09 12:02:19.832 [DEBUG] [ternal.handler.SimaticGenericHandler] - Simatic S7-1200 - initialize. Channels count=534
    2022-03-09 12:02:24.616 [DEBUG] [.simatic.internal.simatic.SimaticTCP] - 192.168.1.30 - open() - connecting...
    ---- first read from PLC
    2022-03-09 12:02:26.132 [DEBUG] [.simatic.internal.simatic.SimaticTCP] - 192.168.1.30 - Reading finished in 16ms. Area=M2-M219
    2022-03-09 12:02:32.436 [INFO ] [e.automation.internal.RuleEngineImpl] - Rule engine started.
    ---- first value PLC changed
    2022-03-09 12:02:33.153 [INFO ] [openhab.event.ItemStateChangedEvent ] - Item 'PLC_OczkoPompaCzasOdliczanie' changed from 507147 to 506152
    2022-03-09 12:02:46.798 [INFO ] [io.openhabcloud.internal.CloudClient] - Connected to the openHAB Cloud service (UUID = 1a790de9-cd89-42a5-b3cf-f6fb306e9bd1, base URL = http://localhost:8080)
    2022-03-09 12:02:55.215 [INFO ] [ab.ui.habpanel.internal.HABPanelTile] - Started HABPanel at /habpanel
    ---- OH start finished

OH start is really slow. There is almost a minute between Simatic binding started and the Rule engine started. 12 seconds between PLC bridge initialization and Rule engine started. After the Rule engine starts, the OH event bus is usually ready (item values are updated). 

At the moment I don't know of a solution other than increasing the binding initialization time.

docbender, 2022-03-09:
I was thinking about the possibility to change binding start-level value. It is set to 80 for all bundles as default. You can check it in karaf console with this command:

    bundle:list

All bundles start in parallel. I tested starting with a higher value and it seems that this should be the solution for your problem.

Try to increase binding start-level from karaf console with this command:

    bundle:start-level org.openhab.binding.simatic 81

grzegorz914, 2022-03-10:
Thanks for the investigation but this doesn’t solve the problem. 
May be use timeout and refresh and push all values  to the oh.core (only one time) after initialization complete?

grzegorz914, 2022-03-10:
Opps the bundle:start-levelwas reset to 80 after restart, so I can not test it in this way.

docbender, 2022-03-23:
Addon loaded from OH addon folder is started first as you can see in console bundle:list command. This can be only affected by increase value of start level. This behavior does not occur in a standard binding installation.

I added binding into marketplace and if it is installed from here, binding is launched last. 

To install binding from marketplace I recommend:
* stop OH
* remove binding from addon folder
* run _openhab-cli clean-cache_ command 
* start OH
* install binding again from marketplace

grzegorz914, 2022-03-30:
Hi,
I can confirm that after restart OH or cycle power off/on the complete system the values are now updated correct but 
the problem stil exist after OH update to new build and first start OH service. After restart services again the values are updated correct.

docbender, 2022-04-05:
This whole is OH core behaviour. Obviously a lot happens in OH after an update, like creating a new cache, resulting in further delays in system startup.
There is no other option besides the ones I already mentioned. At least I don't know of any.

PR #41 (2022-03-08): Ensure initialization completed
Fix #38, fix #39
closed 2022-04-05 by docbender