docbender/openHAB-Simatic #42: Beginner question: how to determine stateAddress and commandAddress (from existing TIA Portal project)

Feestaap, 2022-05-29:
I am a beginner in the PLC world and am trying to transfer an existing TIA Portal project (not my build) to OpenHab.

I wanted to start with a simple light switch, but I am already stuck at finding the right stateAddress and commandAddress for the switch to function in OpenHab.

I have already tried every possible combination for the adresses I could think of, but nothing seems to work.

I have added a screenshot from the TIA Portal for this switch and posted my .things file below.

Can someone please point me in the right direction? 

```
Bridge simatic:bridge:PLC-1200 "PLC "  [ address="xxx.xxx.xx.xxx", rack="0", slot="1"] {
    Thing generic_device plcState "Lights" {
        Channels:
            Type chSwitch: kantoor_achter [ stateAddress="DB9.DBQ0.0", commandAddress="DB9.DBI0.0" ]
}
}
``
![DB9](https://user-images.githubusercontent.com/106469947/170862130-4c3a1827-c83a-4548-8d56-5c88569f3130.PNG)
`

Maddin2020, 2022-05-29:
Hi you must use this Syntax stateAddress="DB9.DBX0.0", commandAddress="DB9.DBX0.0" But i think you better create a new Bit and use it with a OR Gatter on the Input from the Function.

Feestaap, 2022-05-29:
> Hi you must use this Syntax stateAddress="DB9.DBX0.0", commandAddress="DB9.DBX0.0" But i think you better create a new Bit and use it with a OR Gatter on the Input from the Function.

Thank you very much for your reply, but the syntax you suggest unfortunately does not work. It produces the following error: _COMMUNICATION_ERROR
Read data area error (Area=DB9.DBB0-DB9.DBB1, Return code=0x5, Error=the desired address is beyond limit for this PLC})_

gertjanlobbens, 2022-10-22:
I know... old post.
But because many people will struggle with 'simple' things like this i'll show u the most simple way to switch a light with openhab and a Siemens plc.

**In TIA Portal:**
Create a new FB (FB_Flip_Flop)
At the top u define 1 input (In, Bool), 1 InOut (Out, Bool) and 2 statics (Flank, Bool and Memory, Bool).
We only need 2 networks to make this work. The image below shows the entire config for the block.

[image]

Call the FB in your program and fill in the in and output parameters:

[image]

Download the program to your PLC.

**In OpenHab:**
Create a Simatic Bridge, fill in your plc's ip-address. Rack will be always 0 (except for S7-400), slot depends on the type of plc (S7-1200 = 1, S7-300 = 2, ...).

Create a Simatic Thing, in the config fill in the bridge u just made.
On the top go to Channels and click on 'Add Channel'
Give the channel a name and choose On/Off Switch.
For the state address and command address fill in the output of the plc (Q0.0).
Click create.
Click on Add Link to Item... , choose 'Create a new item' and click Link.

You should now have a channel and an item linked to it:
[image]

On your OpenHab page you can now add a toggle card and choose the item you just created.

This is one of the most simple methods to switch a light with OpenHab and a Siemens PLC. I know there are many more options to do this but once you get the concept of it you can do almost anything you like.

**On a side note:**
If you use DB's in your plc (S7-1200 / S7-1500) and like to connect to the items in it you MUST turn off 'optimized access' in the properties of the block!

Right click the DB and go to properties.
Under attributes deactivate 'Optimized block acces'.

[image]

[image]

The X in DB200.DBX0.0 stands for a boolean (0 or 1) value.
for number values u use:
B (byte -> DB200.DBB0)
W (word or integer -> DB200.DBW0)
D (dword or dint or real  -> DB200.DBD0)
For real (also called floating points) you have to put an F in the address in OpenHab -> DB200.DBD0F

Hope this helps someone :-)

Feestaap, 2022-10-24:
I finally had some time to pick up this project again on Saturday, so I could not believe my luck when I saw that night that there was another reply to my question. You're tutorial helped me a lot and finally made things click, so, at last, I am getting somewhere. Thank you so much for taking the time to answer!

docbender, 2022-11-02:
Nice work @gertjanlobbens. 
I would like to put your example in the wiki for others to see.

closed 2022-11-02 by docbender