docbender/openHAB-Simatic #46: O.H. 4.0 (Unit) number channel with command address and unit is not updated

grzegorz914, 2023-05-26:
Hi,

after latest update in O.H. 4.x and support default unit for Number the channels which uses command address and unit are not updated anymore, always return 0. After remove unit all working correct.

[image]

Thanks
Grzegorz

docbender, 2023-06-26:
It is working for me.  

Isn't the problem that OH now displays percentages in the range 0-1? Received values are divided by 100. So OH displays 1, if PLC value is 100.

grzegorz914, 2023-06-26:
Yes, that was the problem, my Channels are 0-100%.

docbender, 2023-06-26:
Set items metadata to unit = %, then OH displays values as 0-100.

closed 2023-06-26 by docbender