docbender/openHAB-Simatic #47: UNKNOWN error on the bridge

Jasper202, 2023-08-11:
I just updated to openhab 4.0.1. On my previous version (3.4) I worked with the simatic bridge to connect to my S7-1500 PLC. I just installed the new simatic binding and I keep having the issure that the simatic bridge stays in state "UNKNOWN" instead of going online. Thanks in advance for the support! Let me know if more information is needed.

Jasper202, 2023-08-12:
update from my end: the issue keeps happening when I installed the binding trough marketplace. When I donwloaded the JAR file and placed it in the addons folder, everything seems to work.

grzegorz914, 2023-08-12:
In the marketplace still the 3.x version, on GitHub 4.x.x beta adapted to OH4

docbender, 2023-08-14:
In the marketplace the OH4 binding was marked as beta. It was only visible when the corresponding option in the OH settings was enabled. Now the binding is marked as stable and is available on the marketplace.

closed 2023-08-14 by docbender