docbender/openHAB-Simatic #49: chNumber - write to the PLC does not work.

hynekprokop, 2024-12-23:
Hi,
Thank you for your work on the binding. I have been using this binding for several years. 
I am running OpenHab 4.3 from the official Docker image. Binding version: [4.0.0;4.9.0) (Updated 14 August 2023 8:42 AM)
**What works as expected:**
chRollershutter, chContact, chSwitch - both directions are ok

**What does not work for me:** 
chNumber - only direction from PLC to OpenHab works. 

When I send **REFRESH command**, I can see OpenHab CommandEvent and also DEBUG message from Simatic Binding, _SimaticGenericHandler.handleCommand(***)_
[image]

When I **change value of the item**, I can see CommandEvent but NO DEBUG message in the log:
[image]

Since the first line in _SimaticGenericHandler.handleCommand(***)_ is logger.debug(), Do I understand it correctly that the problem is that Openhab does not send this event to the Binding? I think I used this feature before (1 or 2 years back) and it was working.
`public void handleCommand(ChannelUID channelUID, Command command) {
        logger.debug("{} - Command {}({}) for channel {}", thing.getLabel(), command, command.getClass(), channelUID);
`

I also try to raise this issue in the OpenHab community forum: [Issue](https://community.openhab.org/t/item-number-change-is-not-updated-in-binding/161036)

docbender, 2024-12-23:
Hi, I tested it with OH4.3 and the last released binding 4.0.1. The data transfer worked fine for me in both directions.

In my case I created a number channel with the same command and status address MB900. To set the value I linked item on a new page to the slider.

hynekprokop, 2024-12-23:
Hi. I tried that just now.

  - id: BlindAngle
    channelTypeUID: simatic:chNumber
    label: Blind Angle
    configuration:
      commandAddress: DB3.DBD104  (I also tried DBW)
      stateAddress: DB3.DBD104

Value gets updated from PLC, but slider does not write value to the PLC.
Also in log, I can see that slider generates CommandEvents, but no DEBUG message from handleCommand:
Same thing happens when I try set the value from the script (_items.getItem("OfficeBlind_angle").sendCommand("655");_).
[image]

I install binding from Add-on store:
Where I can see exact version? I install it one week ago. 
[image]

What type of the item did you use? Before I had Semantic class "Setpoint". Now it looks like this:
[image]

After this change, I have new message in the log :
[image]

Accepted type is coming from the binding?

hynekprokop, 2024-12-24:
Clean installation of OpenHab + Binding works as it should. 
So unfortunately I will have to rebuild my 300+ item / 30+ thing openhab by hand again.

hynekprokop, 2025-01-02:
It happens when I change the channel type. Could you please try it yourself? 
In my case I change the channel from chRollerShutter to chNumber. Then OpenHab stops writing to the PLC. Only deleting everything and reinstalling the binding fixes it.
I think the problem is in the OpenHab, but I was told to confirm it first with the binding author before rasing an issue in the OpenHab repository. Can you please test and confirm?

docbender, 2025-01-09:
Normally it is not possible to switch channel type in the GUI. Do you use text files item definition? Is it where data type change is done?

hynekprokop, 2025-01-11:
I change it using UI -> Thing-> Code tab: YAML text configuration.

docbender, 2025-01-17:
I tested the behavior of openHAB 4.3.1 with Simatic binding after changing the channel type using the code section of the channel configuration.

After changing the channel type, the data type of the linked item must be changed too. Before confirming the change, I was informed that linked item may be broken.

After these changes, binding stops receiving command from openHAB. The handleCommand method is not called. It looks like some internal issue with item configuration. If I create a new item and link it to the same channel, the command from new item is then passed to the binding.

hynekprokop, 2025-01-17:
Thanks for the info. If it continued to work after you created a new item, then it is ok. 
Could you please try to add a new item when it is in "broken state"? Because I was not able to. The screen in the main UI where you select Simatic Thing or Bride to create was empty.

So I think you can close this issue. It looks like an OpenHab problem. 
Thanks for your help.

docbender, 2025-01-19:
In my case GUI still worked normally. Both a broken and a newly created item were assigned to the thing channel at the same time.

closed 2025-01-19 by docbender